Mixiaoxiao/Arduino-HomeKit-ESP8266 #1: Arduino cannot execute

CHNleing, 2020-02-25:
excuse me. I encountered a problem while testing the example. Aruino IDE tells me:
homekit/types.h: No such file or directory
homekit/homekit.h: No such file or directory
homekit/characteristics.h: No such file or directory

I installed through the management library

CHNleing, 2020-02-25:
额。。。没细看 北京人啊，
请教下 我在arduino IDE中打开示例文件进行编译时出现错误。
无法找到homekit相关的文件。
还请帮忙看下，

CHNleing, 2020-02-25:
Found the problem.
I should put arduino_homekit_server.h on top 

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <arduino_homekit_server.h>
#include <homekit/types.h>
#include <homekit/homekit.h>
#include <homekit/characteristics.h>

euler271, 2020-02-26:
Libraries <homekit/types.h>, <homekit/homekit.h> and <homekit/characteristics.h> are not required for the example to work. The should be omitted from simple_led.ino. I recently created a pull request to fix this issue.

Mixiaoxiao, 2020-02-26:
Thanks. I used a little bit intelligent IDE Sloeber(based on eclipse) to build this library and did not meet this issue.

closed 2020-02-26 by Mixiaoxiao