Mixiaoxiao/Arduino-HomeKit-ESP8266 #2: Recently found a conflicting library, causing it to not run properly

CHNleing, 2020-02-26:
[WiFiManager](https://github.com/tzapu/WiFiManager/tree/development#on-demand-configuration-portal)

Mixiaoxiao, 2020-02-26:
The reason is that the `HTTP_METHOD` (eg `HTTP_GET` , `HTTP_POST`) is multi-defined in `ESP8266WebServer` library (used in WiFiManager) and `http_parser.h` of this library. I will change the definitions in `http_parser.h` in next version to make it compatible with `ESP8266WebServer`.

Mixiaoxiao, 2020-02-28:
Hi, please check v1.0.1

CHNleing, 2020-02-28:
> 嗨，请检查v1.0.1

Thank you, I will update now.

CHNleing, 2020-02-28:
I suggest that you can make examples simpler and remove the physical keys. This makes it easier for novices to understand it

euler271, 2020-02-28:
@CHNleing I will work on a simpler example over the weekend and make a pull request for it afterwards.

closed 2020-03-11 by Mixiaoxiao