Mixiaoxiao/Arduino-HomeKit-ESP8266 #7: HomeKit breaks NeoPixel Support

bobmosh, 2020-03-24:
Hey!

Love the library! It works like a charm on its own. Really loving it this far!

However I’ve tried connecting a neopixel and running it through the Adafruit library. Somehow, the setup of HomeKit breaks the neopixel library. The LEDs don’t get the right signals and flicker in weird colors and brightness.

Do you think there is a way to make them both work in one sketch?

CHNleing, 2020-03-31:
You can refer https://github.com/CHNleing/ESP8266HomeKitDome/tree/WS2812B_Color_LED
Strange colors and brightness, mostly caused by insufficient power.
WS2812B requires 0.3W per LED

bobmosh, 2020-04-15:
Thanks! I'll take a look at this and report back to you!

closed 2020-06-10 by Mixiaoxiao