Mixiaoxiao/Arduino-HomeKit-ESP8266 #8: More examples on the way

sorriso93, 2020-03-30:
Hello
do you think you will release more example with more accessories?
Anyway, great job, I'll try to build my project with it (Motion sensor + sht30 sensor)

xfdr0805, 2020-04-27:
me too,good job.  I'll try to build my project with it (sonoff 4ch),tks!

pqhf5kd, 2020-05-08:
Any work on these additional examples?

Mixiaoxiao, 2020-05-11:
On the way... This weekend or next weekend. I am a little busy these days.

------------------------

One sensor example: (main code only)

Step 1. 
Create a .c file, eg "my_accessory.c"
Copy these lines form [temperature_sensor.c](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/temperature_sensor/temperature_sensor.c)
```C
homekit_characteristic_t temperature = ...
homekit_characteristic_t humidity    =  ...
homekit_accessory_t *accessories[] = ....
homekit_server_config_t config = ...
```

Step 2. 
In your sketch (the xxxx.ino file), you can include the .h library for your sensors here (eg. <DHT.h>). 
```C
//write these lines to access the characteristics and config defined in your .c file
extern "C"  homekit_characteristic_t temperature;
extern "C" homekit_characteristic_t humidity;
extern "C" homekit_server_config_t config;

//when you want to report your sensor data: (eg. every 10 seconds)
float temperature_value = dht.readTemperature();
float humidity_value = dth.readHumidity();
temperature.value.float_value = temperature_value; //your temperature data, float
humidity.value.float_value = humidity_value;//your humidity data, float
homekit_characteristic_notify(&temperature, HOMEKIT_FLOAT(temperature_value));//report the value to iOS
homekit_characteristic_notify(&humidity, HOMEKIT_FLOAT(humidity_value));//report the value to iOS

```

NOTE:
The homekit library is mostly written in C, you can define your  accessory/service/characteristic in a .c file, since the library provides convenient Macro (c only, cpp can not  compile) to do this. (But it is possible to do this in cpp or ino, just not so convenient).

Generally, the Arduino Libraries (eg, sensors and ws2812) are written in cpp, you can include them in a .ino or a .cpp file (but can NOT in .c). A .ino is a .cpp indeed.

You can define some variables in a .c file, eg `int my_value = 1;`, and you can access this variable in a .ino or a .cpp by writing `extern "C" int my_value;`.

So, if you want use this homekit library and other Arduino Libraries together, 

1. define your homekit accessory/service/characteristic in a .c file
2. in your .ino, include some Arduino Libraries and you can use them normally
3. in your .ino, write `extern "C"  homekit_characteristic_t your_characteristic;` to access the characteristic defined in your .c file
4. in your .ino, write your logic code (eg. read sensors) and report your data to homekit by writing `your_characteristic.value.xxxx_value = some_data;  homekit_characteristic_notify(.., ...)` 

done.

xfdr0805, 2020-05-12:
Here is the example demo url  [https://github.com/maximkulkin/esp-homekit-demo](url)

Mixiaoxiao, 2020-05-14:
A new example is uploaded, check out the folder `example/Example01_TemperatureSensor`.
More examples are on the way.

closed 2020-05-25 by Mixiaoxiao