Mixiaoxiao/Arduino-HomeKit-ESP8266 #9: Random connection loss

alexmrtn, 2020-04-07:
Hi, 

First of all thanks for this porting! I tried the original esp-homekit but your work here made the use of this library much more simple!

I noticed that sometime all of my devices (not always at the same time) show that they aren't reachable. I was able to reproduce the behavior with you simple led example, and here the logs:
```
16:40:10.599 > >>> [ 748449] HomeKit: [Client 1073683172] Found pairing with C494D193-4EBE-47CB-8422-5E03001BDBC1
16:40:10.615 > >>> [ 748477] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
16:40:10.641 > >>> [ 749247] HomeKit: [Client 1073683172] Verification successful, secure session established
16:40:11.411 > >>> [ 749260] HomeKit: Free heap: 35512
16:40:11.419 > >>> [ 749274] HomeKit: [Client 1073683172] Get Accessories
16:40:11.435 > heap: 36752, sockets: 4
16:40:14.012 > heap: 36752, sockets: 4
16:40:19.013 > heap: 36752, sockets: 4
16:40:24.014 > heap: 36752, sockets: 4
16:40:29.015 > heap: 36752, sockets: 4
16:40:34.016 > heap: 36752, sockets: 4
16:40:39.017 > heap: 36752, sockets: 4
16:40:44.018 > heap: 37088, sockets: 4
16:40:49.019 > >>> [ 788020] HomeKit: [Client 1073683380] Get Characteristics
16:40:50.182 > heap: 36752, sockets: 4
16:40:54.020 > heap: 36944, sockets: 4
16:40:59.021 > heap: 37088, sockets: 4
16:41:04.022 > heap: 36752, sockets: 4
16:41:09.023 > heap: 36752, sockets: 4
16:41:14.024 > heap: 37096, sockets: 4
16:41:19.025 > heap: 36752, sockets: 4
16:41:24.026 > heap: 36752, sockets: 4
16:41:29.028 > >>> [ 827280] HomeKit: [Client 1073679716] Disconnected!
16:41:29.441 > >>> [ 827288] HomeKit: [Client 1073679716] Closing client connection
16:41:29.451 > heap: 38368, sockets: 3
16:41:34.028 > >>> [ 836298] HomeKit: [Client 1073683380] Disconnected!
16:41:38.459 > >>> [ 836306] HomeKit: [Client 1073683380] Closing client connection
16:41:38.469 > heap: 39936, sockets: 2
16:41:39.030 > >>> [ 838861] HomeKit: [Client 1073683172] Disconnected!
16:41:41.022 > >>> [ 838869] HomeKit: [Client 1073683172] Closing client connection
16:41:41.032 > >>> [ 840085] HomeKit: [Client 1073683212] Disconnected!
16:41:42.245 > >>> [ 840092] HomeKit: [Client 1073683212] Closing client connection
16:41:42.255 > heap: 43096, sockets: 0
16:41:44.031 > heap: 43440, sockets: 0
16:41:49.032 > heap: 43440, sockets: 0
16:41:54.033 > heap: 43440, sockets: 0
16:41:59.034 > heap: 43440, sockets: 0
16:42:04.035 > heap: 43440, sockets: 0
16:42:09.036 > heap: 43096, sockets: 0
16:42:14.037 > heap: 43096, sockets: 0
16:42:19.038 > heap: 43096, sockets: 0
16:42:24.039 > heap: 43096, sockets: 0
16:42:29.040 > heap: 43096, sockets: 0
16:42:34.041 > heap: 43096, sockets: 0
16:42:39.042 > heap: 43096, sockets: 0
16:42:44.043 > heap: 43096, sockets: 0
16:42:49.044 > heap: 43096, sockets: 0
16:42:54.045 > heap: 43096, sockets: 0
16:42:59.046 > heap: 43096, sockets: 0
16:43:04.047 > heap: 43096, sockets: 0
16:43:09.048 > >>> [ 928082] HomeKit: Got new client: local 192.168.1.60:5556, remote 192.168.1.43:51900
16:43:10.250 > >>> [ 928096] HomeKit: [Client 1073683380] Pair Verify Step 1/2
16:43:10.259 > >>> [ 928423] HomeKit: Free heap: 40120
16:43:10.583 > >>> [ 928458] HomeKit: [Client 1073683380] Pair Verify Step 2/2
16:43:10.619 > >>> [ 928468] HomeKit: [Client 1073683380] Found pairing with C494D193-4EBE-47CB-8422-5E03001BDBC1
16:43:10.636 > >>> [ 928496] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
16:43:10.661 > >>> [ 929266] HomeKit: [Client 1073683380] Verification successful, secure session established
16:43:11.432 > >>> [ 929279] HomeKit: Free heap: 40912
16:43:11.439 > >>> [ 929296] HomeKit: [Client 1073683380] Get Accessories
16:43:11.458 > >>> [ 929344] HomeKit: Got new client: local 192.168.1.60:5556, remote 192.168.1.53:49812
16:43:11.511 > >>> [ 929357] HomeKit: [Client 1073682268] Pair Verify Step 1/2
16:43:11.521 > >>> [ 929684] HomeKit: Free heap: 36032
16:43:11.845 > >>> [ 929714] HomeKit: [Client 1073682268] Pair Verify Step 2/2
16:43:11.876 > >>> [ 929724] HomeKit: [Client 1073682268] Found pairing with C494D193-4EBE-47CB-8422-5E03001BDBC1
16:43:11.892 > >>> [ 929752] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
16:43:11.917 > >>> [ 930522] HomeKit: [Client 1073682268] Verification successful, secure session established
16:43:12.688 > >>> [ 930535] HomeKit: Free heap: 38832
16:43:12.695 > >>> [ 930582] HomeKit: [Client 1073682268] Get Accessories
16:43:12.744 > >>> [ 930675] HomeKit: [Client 1073682268] Update Characteristics
16:43:12.838 > >>> [ 931430] HomeKit: Got new client: local 192.168.1.60:5556, remote 192.168.1.12:56080
16:43:13.597 > >>> [ 931444] HomeKit: Got new client: local 192.168.1.60:5556, remote 192.168.1.19:62318
16:43:13.611 > >>> [ 931457] HomeKit: [Client 1073687620] Pair Verify Step 1/2
16:43:13.620 > >>> [ 931784] HomeKit: Free heap: 34200
16:43:13.946 > >>> [ 931790] HomeKit: [Client 1073682300] Pair Verify Step 1/2
16:43:13.953 > >>> [ 932117] HomeKit: Free heap: 33576
16:43:14.277 > heap: 34336, sockets: 4
16:43:14.284 > >>> [ 932254] HomeKit: [Client 1073682300] Pair Verify Step 2/2
16:43:14.415 > >>> [ 932264] HomeKit: [Client 1073682300] Found pairing with C494D193-4EBE-47CB-8422-5E03001BDBC1
16:43:14.432 > >>> [ 932291] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
16:43:14.457 > >>> [ 933062] HomeKit: [Client 1073682300] Verification successful, secure session established
16:43:15.227 > >>> [ 933075] HomeKit: Free heap: 34400
16:43:15.235 > >>> [ 933175] HomeKit: [Client 1073682300] Get Accessories
16:43:15.337 > >>> [ 933679] HomeKit: [Client 1073683380] Update Characteristics
16:43:15.843 > >>> [ 935625] HomeKit: [Client 1073687620] Pair Verify Step 2/2
16:43:17.787 > >>> [ 935635] HomeKit: [Client 1073687620] Found pairing with C494D193-4EBE-47CB-8422-5E03001BDBC1
16:43:17.803 > >>> [ 935663] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
16:43:17.828 > >>> [ 936433] HomeKit: [Client 1073687620] Verification successful, secure session established
16:43:18.599 > >>> [ 936446] HomeKit: Free heap: 35896
16:43:18.606 > >>> [ 936647] HomeKit: [Client 1073687620] Get Accessories
16:43:18.810 > heap: 36704, sockets: 4
```

It seems that suddenly the device lost all the connected clients. It is something you have seen before? Any idea of what could cause this?

Thanks,

Mixiaoxiao, 2020-04-07:
Hi, it true that “suddenly the device lost all the connected clients”, but I have no idea about the reason at present. It seems that you have at lease 4 iOS devices connected to the ESP8266. I only have 1 or 2 iOS devices (one iPhone and one MAC, no iPad or HomePod as HUB) in my hand. Nevertheless I will try to figure out this error. 

Another important thing is the state of iOS. Did it show `NOT reachable` when `disconnected` (after the log of 16:41:42.245) and become `reachable` when `reconnected` (after 16:43:18.606)? Or maybe it always be `NOT reachable` after `disconnected` and can NOT recover?

alexmrtn, 2020-04-07:
Yes I have an AppleTV that I use as hub, iPhone and two Mac (that are the four devices). They connects to the device with I guess the iCloud sync. I suppose iCloud share the client key among all of them.

It is hard to tell as it is random behavior, but I'll try to figure out. The full log is [here](https://gist.github.com/alexmrtn/b932aa3fa47509ffefe646e4e3261f0c) if it helps.

alexmrtn, 2020-04-07:
Ok got it, the iOS client shows the `NOT reachable` *before* the ESP logs `disconnected`. It recover automatically few minutes after (no need to reboot the device). 

It is also strange that all the device disconnects at the same time. By all devices I mean all that run the Arduino-Homekit-ESP8266 library, other devices are not affected. One possible reason could be that a broadcast message from the iOS device, is not properly handled on the ESPs. 

I'll try to understand the debug logs using the `HOMEKIT_DEBUG` build flag.

Mixiaoxiao, 2020-04-08:
> One possible reason could be that a broadcast message from the iOS device, is not properly handled on the ESPs.

Maybe something in Apple's HAP is not implemented properly by the MDNS library.

alexmrtn, 2020-04-11:
It seems that there is indeed something related to MDNS. I monitored the announcements using the `dns-sd -B _hap` command here some logs:

```
17:02:58.917  Rmv        1   5 local.               _hap._tcp.           Switch_AD6D41
17:02:58.918  Rmv        1   5 local.               _hap._tcp.           Switch_9B0715
17:02:58.918  Rmv        1   5 local.               _hap._tcp.           Switch_CA9757
17:02:58.918  Rmv        1   5 local.               _hap._tcp.           Switch_AD93EF
17:02:58.918  Rmv        1   5 local.               _hap._tcp.           Switch_9BB57F
17:02:58.918  Rmv        1   5 local.               _hap._tcp.           Switch_9B9C08
17:02:58.918  Rmv        0   5 local.               _hap._tcp.           Switch_9B0983
17:03:33.049  Add        2   5 local.               _hap._tcp.           Switch_810E0D
17:04:26.887  Add        3   5 local.               _hap._tcp.           Switch_9B0715
17:04:26.887  Add        3   5 local.               _hap._tcp.           Switch_9B0983
17:04:26.887  Add        3   5 local.               _hap._tcp.           Switch_CA9757
17:04:26.887  Add        3   5 local.               _hap._tcp.           Switch_9BB57F
17:04:26.887  Add        3   5 local.               _hap._tcp.           Switch_AD6D41
17:04:26.887  Add        3   5 local.               _hap._tcp.           Switch_9B9C08
17:04:26.887  Add        3   5 local.               _hap._tcp.           ESP8266_LED_7B7A75
17:04:26.887  Add        2   5 local.               _hap._tcp.           Switch_AD93EF
17:06:12.262  Rmv        1   5 local.               _hap._tcp.           Switch_9B0715
17:06:12.262  Rmv        1   5 local.               _hap._tcp.           Switch_9B0983
17:06:12.262  Rmv        1   5 local.               _hap._tcp.           Switch_CA9757
17:06:12.262  Rmv        1   5 local.               _hap._tcp.           Switch_9BB57F
17:06:12.262  Rmv        1   5 local.               _hap._tcp.           Switch_AD6D41
17:06:12.263  Rmv        1   5 local.               _hap._tcp.           Switch_9B9C08
17:06:12.263  Rmv        1   5 local.               _hap._tcp.           ESP8266_LED_7B7A75
17:06:12.263  Rmv        1   5 local.               _hap._tcp.           Switch_AD93EF
17:06:12.263  Rmv        0   5 local.               _hap._tcp.           Switch_810E0D
17:06:30.821  Add        2   5 local.               _hap._tcp.           Switch_810E0D
17:07:38.081  Add        2   5 local.               _hap._tcp.           ESP8266_LED_7B7A75
17:09:03.841  Rmv        0   5 local.               _hap._tcp.           Switch_810E0D
17:09:12.256  Rmv        0   5 local.               _hap._tcp.           ESP8266_LED_7B7A75
```

Don't know why there is the removing of the services periodically. iOS device shows `Not Reachable` after the the service is removed. Here what it happen:

1. MDNS Service is removed
2. iOS client shows `Not reachable` 
3. TCP socket timeout
4. Device logs `Disconnected`

It doesn't come from the dynamic TXT service `"sf"` as I tried to use a static one, same behavior.

Continuing the investigation...

emelianov, 2020-05-06:
@alexmrtn ,
I've resolved the same problem by periodacally calling
```c
MDNS.announce();
```
Every 5 seconds call works fine for me.

Mixiaoxiao, 2020-05-07:
I think this issue is caused by MDNS with a high probability.
After building some projects and working on ESP32's Arduino HomeKit, I found that the iOS will disconnect immediately (or become unreachable) in these cases:
- Something wrong with MDNS: no response, or TXT changed or is wrong
- Something wrong with characteristics: error format, error value(ie, out of range)

Another important thing is that the iOS "think" the accessory is reachable or not is highly depends on the MDNS broadcast.

chrisos, 2020-05-07:
Hi,

First of thanks to @Mixiaoxiao for all the great work youve done here. I think everyone appreciates it very much :)

Second, @emelianov, thanks for the advice! I had a similar problem, were the device would disconnect whenever the Iphone is about 20s in the lock screen. I added the MDNS.announce() in the arduino_homekit_loop() in arduino_homekit_server.cpp and it works like a charm.

All of you, keep up the great work, and I am looking forward to more examples!

Best regards,

Chrisos

emelianov, 2020-05-07:
@Mixiaoxiao you are right. I've facing this MDNS problem (lost device name resolution) when experimenting with ESP IDF-SDK and Non-OS SDK both.
@chrisos putting `MDSN.announce` will cause it calls too frequently and produce lot of broadcast traffic on network. So i suggest to use less frequent calls schedule.

chrisos, 2020-05-07:
@emelianov Thanks for the Tip, will do! Have 5s at the moment but will test were a proper limit is.

chrisos, 2020-05-07:
Hi, 

@emelianov, contrary to my last post I did not have a delay implemented. I thought, but I did misinterpret the code. 

I tested the optimal time period for updating MDNS.announce() from my previous post. I found 1500ms optimal. With this delay the Iphone never looses complete connection and it should reduce the amount of traffic on the network.

Best regards

alexmrtn, 2020-05-08:
Thanks all for the feedback, I'll try to add `MDNS.announce()` call in the main loop and see if it solve the problem!

lawrencepn, 2020-05-13:
> @alexmrtn ,
> I've resolved the same problem by periodacally calling
> 
> ```c
> MDNS.announce();
> ```
> 
> Every 5 seconds call works fine for me.

This worked for me too. thanks for the tip.

alexmrtn, 2020-09-23:
I finally found some time to get back to this, with the `MDNS.announce()` all is working great! Thanks all :)

closed 2020-09-23 by alexmrtn

CommyD, 2020-10-03:
hi, can someone please let me know how to use/ where to write MDNS.announce()? a printscreen would really help :)
thansk

v0r73x, 2020-11-23:
I have same question - how to set it up? If i use it only by putting it into the homekit loop the comiler says : 'MDNS' was not declared in this scope

emelianov, 2020-11-23:
```c
#include <ESP8266mDNS.h>
```

#71

v0r73x, 2020-11-23:
Hi emelianov,

can i use this library as well on my espwroom32? or should i use there another one?
thanks for your help - i am starting new into this topic.

emelianov, 2020-11-23:
For ESP32 it should be

```c
#include <ESPmDNS.h>
```

Code from #71 will not work for ESP32.

tk97tk, 2021-08-28:
@emelianov Do you know the equivalent method to MDNS.announce() in the <ESPmDNS.h> library?