Mixiaoxiao/Arduino-HomeKit-ESP8266 #10: Multiple HomeKit Device pairing issue

bobmosh, 2020-04-15:
Hey!

Again awesome work on the library!
So far I had only one device in my network. When I try to add a different one I am not able due to the warning message: "Found admin pairing with 6AE7DA31-6F0A-4B0F-9E81-FECEBC2F6FA4, disabling pair setup"

Full Print:
```
>>> [   4341] HomeKit: Update the CPU to run at 160MHz
>>> [   4346] HomeKit: Starting server
>>> [   4349] HomeKit: Using existing accessory ID: FC:FA:56:16:CA:F8
>>> [   4356] HomeKit: Found admin pairing with 6AE7DA31-6F0A-4B0F-9E81-FECEBC2F6FA4, disabling pair setup
>>> [   4365] HomeKit: Configuring MDNS
>>> [   4370] HomeKit: MDNS.begin: Rise Wecker_E14627, IP: 192.168.0.145
>>> [   4380] HomeKit: Init server over
>>> [   4384] HomeKit: Free heap: 43744
```

Seems like both my devices have the same device id. How is this possible?
How can I change the device ID for the second ESP?
I do have two physically different devices so this confuses me.

Cheers,
Ferdinand

Mixiaoxiao, 2020-04-15:
Key points:
- One ESP (one accessory) can **ONLY** paired with **ONE** iOS device (this is the "admin pairing").
- The paired ESP can be controlled by other iOS devices which have the same iCloud ID with the "admin pairing" iOS device.
- The paired ESP will refuse to pair again (shown by the log "Found admin pairing with xx, disabling pair setup").

You should:
- Use the HOME app on the admin pairing iOS to remove the ESP accessory when ESP is **ONLINE**. Notice that the "pairing removing" progress requires the network connection between iOS and ESP.
- If you have removed the pairing in HOME app when ESP is **OFFLINE**, then "pairing removing" progress will not be finished permanently. In this case, you need to remove the pairing on ESP manually: call `homekit_storage_reset()` function to remove all information saved for HomeKit (eg. trigger by a long-press button event, see the example); Or erase full flash for your ESP and then re-flash your ESP.  

For more detail, you can refer to the Apple's HAP document (HAP-Specification-Non-Commercial-Version.pdf, [https://developer.apple.com/homekit/](https://developer.apple.com/homekit/)).

spxnr, 2020-05-18:
I am having (perhaps) the same issue where I have **2** ESP8266 flashed with the same image (new Switch example) but only 1 of the 2 ESP8266 are showing up in the Home app. I tried incrementing the serial number, but it still didn't show up. I changed the "Name" characteristic (leaving everything else the same) and then the 2nd ESP8266 was discoverable on the Home app.

Mixiaoxiao, 2020-05-25:
Will be closed in a week since there is no following response.

closed 2020-06-01 by Mixiaoxiao

salomoneslait, 2022-03-21:
> I am having (perhaps) the same issue where I have **2** ESP8266 flashed with the same image (new Switch example) but only 1 of the 2 ESP8266 are showing up in the Home app. I tried incrementing the [...]

Hello, I’m having the same problem do you resolved it ?

brunoaduarte, 2023-08-09:
Hi! Why isn't it possible to pair 2 different iOS devices (different iClouds) ?

brunoaduarte, 2023-08-09:
Hi! Why isn't it possible to pair 2 different iOS devices (different iClouds) to 1 device ?