Mixiaoxiao/Arduino-HomeKit-ESP8266 #11: 使用esp-01s出现连接失败

xiaoqianliya, 2020-04-19:
使用esp-01s编译示例进去，出现了错误！
[    112] HomeKit: Starting server
!!! [    116] HomeKit: Failed to read HomeKit storage magic
>>> [    121] HomeKit: Formatting HomeKit storage at 0x3fb000
!!! [    126] HomeKit: Failed to erase HomeKit storage
>>> [    131] HomeKit: Resetting HomeKit storage
!!! [    136] HomeKit: Failed to reset HomeKit storage
>>> [    141] HomeKit: Generated new accessory ID: 2D:C4:6B:9E:03:41
!!! [    147] HomeKit: Failed to write accessory ID to HomeKit storage
>>> [    205] HomeKit: Generated new accessory key
!!! [    210] HomeKit: Failed to write accessory key to HomeKit storage
>>> [    217] HomeKit: Preiniting pairing context
使用iphone连接是否出现的错误！
!!! [  39382] HomeKit: Failed to compact HomeKit storage: sector data read error
!!! [  39389] HomeKit: Failed to write pairing info to HomeKit storage: max number of pairings
!!! [  39397] HomeKit: [Client 1073682852] Failed to store pairing (code -2)
>>> [  39634] HomeKit: [Client 1073682852] Disconnected!

求解决办法，谢谢！

xiaoqianliya, 2020-04-19:
@Mixiaoxiao 求解决，谢谢！

Mixiaoxiao, 2020-04-19:
0x3fb000地址(约为3.98MB的位置)擦写失败，esp01大部分都是1MB的flash。所以问题就在于你按4MB的flash编译的固件烧写到了只有1MB的esp01上。

xiaoqianliya, 2020-04-19:
> 0x3fb000地址（大约3.98MB的位置）擦写失败，esp01大部分都是1MB的闪存。所以问题就在于你按4MB的闪存编译的固件烧写到了只有1MB的esp01上。

@Mixiaoxiao 超级感谢你~~~我把flash size改成1m的就成功配对了！！！超开心，我目前正在制作一个关于这个库的教程，我制作完可以发到酷安这个平台分享吗？会标注好github链接的
再一次感谢你~~

closed 2020-04-20 by xiaoqianliya

DamoclesJ, 2021-07-26:
编译烧写完，家庭 能搜到开关却连不上，求大神解决办法！
WiFi connected, IP: 192.168.43.38
>>> [   5107] HomeKit: Formatting HomeKit storage at 0xfb000
>>> [   5189] HomeKit: Update the CPU to run at 160MHz
>>> [   5194] HomeKit: Starting server
>>> [   5198] HomeKit: Resetting HomeKit storage
>>> [   5202] HomeKit: Formatting HomeKit storage at 0xfb000
>>> [   5282] HomeKit: Generated new accessory ID: E0:4C:25:06:64:B6
>>> [   5341] HomeKit: Generated new accessory key
>>> [   5346] HomeKit: Preiniting pairing context
>>> [   5351] HomeKit: Using user-specified password: 111-11-111
>>> [   5372] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  12058] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [  15475] HomeKit: Preinit pairing context success
>>> [  15480] HomeKit: Configuring MDNS
>>> [  15485] HomeKit: MDNS begin: Switch, IP: 192.168.43.38
>>> [  15492] HomeKit: Init server over
Free heap: 41488, HomeKit clients: 0
Free heap: 42016, HomeKit clients: 0
Free heap: 42016, HomeKit clients: 0
Free heap: 42016, HomeKit clients: 0
Free heap: 42040, HomeKit clients: 0

ryanhsu0307, 2022-06-03:
@xiaoqianliya 請問怎麼修改flash size改成1m？謝謝