Mixiaoxiao/Arduino-HomeKit-ESP8266 #12: PSA: Use 2.4Ghz for adding devices

adamb3ll, 2020-05-07:
Not a bug with your library but I wanted to highlight an issue I was having.

I was adding this library to some old projects I have and slowly building it up ensuring it was working.  Everything was going smoothly until I came back to it a day later and I was having huge issues successfully adding my ESP device.  After many wasted hours debugging I narrowed it down to the fact I have a dual band router and my iOS device was connected to 5Ghz.

I created a guest network (with password) on my router (with sufficient access to my network) and ensured it was 2.4Ghz.  As long as I kept my iPhone connected to the 2.4Ghz network, adding my device worked flawlessly.

I just wanted to highlight the issue for any others who may go near crazy.

Keep up the great work!

Mixiaoxiao, 2020-05-07:
You mean:
The ESP connected to 2.4GHz WiFi and the iOS connected to 5GHz (with the same router), but the HomeKit can not be paired.

If so, maybe there is an issue on your router. That is your router's 2.4GHz and 5GHz are of **separated** networks, not **one** network. You can try changing your router's settings. But I am not sure about this.

adamb3ll, 2020-05-08:
Correct, ESP connected to 2.4,Ghz and iOS and 5Ghz on the same router and trouble pairing.  Once paired I can switch my phone back to my main network from my guest network where it will select 5Ghz again and I can see the added devices no problem.

They are definitely one network.  I can separate them but they are presently not.

Some information I got that led me to my 2.4Ghz test was this:

https://jlian.co/post/reliable-homekit/

Notably this quote:

_**Bonjour services can sometimes work poorly across bands**_

_The real revelation for me came from this reddit post. Essentially, routers can sometimes do wonky stuff when dealing with Bonjour/mDNS (this is what your phone uses to find HomeKit devices in your network) services across the 2.4GHz and 5GHz bands. Armed with this knowledge I eventually found that this is a fairly obscure issue that has been plaguing people for years, especially when it came to wireless printing (AirPrint). This post really drove home the concept for me. Basically, if you can’t see your HomeKit devices using this app, then your router is probably doing something to block Bonjour/mDNS traffic between the 5GHz and 2.4GHz bands._

So yes, it could certainly be an issue on my router but to be honest I'm happy for the short term adding devices via my guest network.  But wanted to share my experience in case anyone suffered the same. I'm happy to close the thread.

closed 2020-05-15 by Mixiaoxiao