Mixiaoxiao/Arduino-HomeKit-ESP8266 #14: I am unable to understand the examples.

gilshahar7, 2020-05-08:
Hi, i have compiled the simplest_led example, it works and all.

But now i want to implement it into my own stuff, i could not understand how the you define an accessories using `homekit_accessory_t *accessories[] = ...`

And how you define what functions get called when the button is on/off, in the example there is a function called `led_toggle();` but i cant find where in the code you define it to run..

anyways the examples are very messy, there are not much comments on the code.

Hope someone can help me out here.

pqhf5kd, 2020-05-11:
I'm also hoping for some more examples. I'd like to read data from a sensor connected to an ESP8266 via serial and report that back to HomeKit

lawrencepn, 2020-05-13:
Took a dive into the code hoping some logic thinking will see me through but I just don't understand how the examples were ported from the original ones. Some clarity on how we can expand on the examples would be appreciated.

eokgnah, 2020-05-14:
A simple Button example would be nice as well.
Would like to add my door-opener with a relais to homekit.
-> homekit_accessory_category_switch sounds like the right one.
Or is a switch bi-stable? Then i would need a "button", but i dont see a category named button.
Thanks in advance.

Mixiaoxiao, 2020-05-14:
> A simple Button example would be nice as well.
> Would like to add my door-opener with a relais to homekit.
> -> homekit_accessory_category_switch sounds like the right one.
> Or is a switch bi-stable? Then i would need a "button", but i dont see a category named button.
> Thanks in advance.

Check out the new example "Example02_Switch".

eokgnah, 2020-05-15:
> Check out the new example "Example02_Switch".

Yes, it works, but it is not exactly what i need.
Now i try to modify the code to just burn the led as long i press the button. If i release the button the led sould go out as well.
Not a bi-stable switch, but a mono-stable push button.
Or after switching on it should switch off by itself after 500ms.

gilshahar7, 2020-05-15:
Thank you @Mixiaoxiao, i saw the added examples and comments. i think i am starting to understand the flow of the code.

i will try to make something with it :)

closed 2020-05-18 by Mixiaoxiao