Mixiaoxiao/Arduino-HomeKit-ESP8266 #15: temperature value error

mcsrgx, 2020-05-11:
Hi, I use DS1820 temperature sensor, when  temperature_value = -1, -2, -3...... and 100, 101, 102....... accessory starting not to response,
so accessory works only with values from 0 to 100

in my .c file code:
float temperature_value = -10;
temperature.value.float_value = temperature_value;
homekit_characteristic_notify(&temperature, HOMEKIT_FLOAT(temperature_value));

any thoughts?
Full example you can find on my github
And thank you for your library very much

emelianov, 2020-05-12:
```c
homekit_characteristic_t temp = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0.00, .getter=f);
...
// Somewhere in initialisation code
  *temp.min_value = -200;
  *temp.max_value = 200;
```

Mixiaoxiao, 2020-05-12:
This is a feature, not a bug.
The "Current Temperature" is defined in the Apple's HAP section 9.35.
The Minimum Value is 0, and the Maximum Value is 100, if your value is out of range, the iOS will reject it and think your accessory has something wrong ( not response).

You can get the "HAP-Specification-Non-Commercial-Version.pdf" here [https://developer.apple.com/support/homekit-accessory-protocol/](https://developer.apple.com/support/homekit-accessory-protocol/)

mcsrgx, 2020-05-12:
Thanks a lot! Every thing works fine
I only needed to add two strings

float temperature_value = t;
temperature.value.float_value = temperature_value;
*temperature.min_value = -200;
*temperature.max_value = 200;
homekit_characteristic_notify(&temperature, HOMEKIT_FLOAT(temperature_value));

But there is only question. Step value shown in home app is 0.5, but in HAP it is 0.1 and my sensor step value is 0.01.
Some thoughts? Why it is 0.5 in home app?

Mixiaoxiao, 2020-05-12:
This also is a feature.
The step of temperature value in iOS HOME App is 0.5 (I have never find a method to modify it).
You can try the "Eve for HomeKit" in App Store, where the temperature value is shown of step 0.1.

closed 2020-05-18 by mcsrgx

louisdcmr, 2022-02-12:
Hello, can I have the code because I plan to make a thermometer also with a ds18b20 but I can't do it…

purpln, 2022-02-14:
@louisdcmr https://github.com/purpln/homekit/tree/master/examples/sensor_temperature

jockethebastard, 2024-03-16:
> @louisdcmr https://github.com/purpln/homekit/tree/master/examples/sensor_temperature

Would you mind sharing this again? I'm having such a hard time configuring the temperature values. The sensor shows up in HomeKit, no problem there, but the values changes randomly

purpln, 2024-03-18:
> this

- [HomeKit](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266)
- [DallasTemperature](https://github.com/milesburton/Arduino-Temperature-Control-Library)
- [OneWire](https://github.com/PaulStoffregen/OneWire)

### temperature.c
```c
#include <Arduino.h>
#include <homekit/types.h>
#include <homekit/homekit.h>
#include <homekit/characteristics.h>
#include <stdio.h>
#include <port.h>

#define ACCESSORY_NAME  ("temperature")
#define ACCESSORY_SN  ("41745062")
#define ACCESSORY_MANUFACTURER ("homekit")
#define ACCESSORY_MODEL  ("esp8285")

homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, ACCESSORY_NAME);
homekit_characteristic_t serial_number = HOMEKIT_CHARACTERISTIC_(SERIAL_NUMBER, ACCESSORY_SN);
homekit_characteristic_t temperature = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0);
float temperature_value;
extern float t;

void temp_update() {
  temperature_value = t;
  temperature.value.float_value = temperature_value;
  *temperature.min_value = -200;
  *temperature.max_value = 200;
  homekit_characteristic_notify(&temperature, HOMEKIT_FLOAT(temperature_value));
  delay(1000);
}

void temperature_sensor_identify(homekit_value_t _value) {
  printf("Temperature sensor identify\n");
}

homekit_accessory_t *accessories[] = {
  HOMEKIT_ACCESSORY(.id = 1, .category = homekit_accessory_category_thermostat, .services=(homekit_service_t*[]){
    HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
      &name,
      HOMEKIT_CHARACTERISTIC(MANUFACTURER, ACCESSORY_MANUFACTURER),
      &serial_number,
      HOMEKIT_CHARACTERISTIC(MODEL, ACCESSORY_MODEL),
      HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
      HOMEKIT_CHARACTERISTIC(IDENTIFY, temperature_sensor_identify),
      NULL
    }),
    HOMEKIT_SERVICE(TEMPERATURE_SENSOR, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "temperature sensor"),
      &temperature,
      NULL
    }),
    NULL
  }),
  NULL
};

homekit_server_config_t config = {
  .accessories = accessories,
  .password = "111-11-111",
};
```

### sensor_temperature.ino
```c++
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <arduino_homekit_server.h>
#define SIMPLE_INFO(fmt, ...)   printf_P(PSTR(fmt "\n") , ##__VA_ARGS__);
const char *ssid = "ssid";
const char *password = "password";

#include <Arduino.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#define term_vcc 2
#define term_gnd 4
#define ONE_WIRE_BUS 0
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
float t = 0.0;

void setup() {
  Serial.begin(115200);
  Serial.setRxBufferSize(32);
  Serial.setDebugOutput(false);
  sensors.begin();
  pinMode(term_vcc, OUTPUT);
  pinMode(term_gnd, OUTPUT);
  
  WiFi.mode(WIFI_STA);
  WiFi.persistent(false);
  WiFi.disconnect(false);
  WiFi.setAutoReconnect(true);
  WiFi.begin(ssid, password);

  SIMPLE_INFO("");
  SIMPLE_INFO("SketchSize: %d", ESP.getSketchSize());
  SIMPLE_INFO("FreeSketchSpace: %d", ESP.getFreeSketchSpace());
  SIMPLE_INFO("FlashChipSize: %d", ESP.getFlashChipSize());
  SIMPLE_INFO("FlashChipRealSize: %d", ESP.getFlashChipRealSize());
  SIMPLE_INFO("FlashChipSpeed: %d", ESP.getFlashChipSpeed());
  SIMPLE_INFO("SdkVersion: %s", ESP.getSdkVersion());
  SIMPLE_INFO("FullVersion: %s", ESP.getFullVersion().c_str());
  SIMPLE_INFO("CpuFreq: %dMHz", ESP.getCpuFreqMHz());
  SIMPLE_INFO("FreeHeap: %d", ESP.getFreeHeap());
  SIMPLE_INFO("ResetInfo: %s", ESP.getResetInfo().c_str());
  SIMPLE_INFO("ResetReason: %s", ESP.getResetReason().c_str());
  INFO_HEAP();
  homekit_setup();
  INFO_HEAP();
}

void loop() {
  homekit_loop();
}

extern "C" homekit_server_config_t config;
extern "C" homekit_characteristic_t name;
extern "C" void temp_update();

void homekit_setup() {
  //homekit_storage_reset();
  uint8_t mac[WL_MAC_ADDR_LENGTH];
  WiFi.macAddress(mac);
  int name_len = snprintf(NULL, 0, "%s_%02X%02X%02X", name.value.string_value, mac[3], mac[4], mac[5]);
  char *name_value = (char*) malloc(name_len + 1);
  snprintf(name_value, name_len + 1, "%s_%02X%02X%02X", name.value.string_value, mac[3], mac[4], mac[5]);
  name.value = HOMEKIT_STRING_CPP(name_value);
  arduino_homekit_setup(&config);
}

void homekit_loop() {
  temperatures();
  temp_update();
  arduino_homekit_loop();
  static uint32_t next_heap_millis = 0;
  uint32_t time = millis();
  if (time > next_heap_millis) {
    SIMPLE_INFO("heap: %d, sockets: %d",
        ESP.getFreeHeap(), arduino_homekit_connected_clients_count());
    next_heap_millis = time + 5000;
  }
}

void temperatures() {
  digitalWrite(term_gnd, LOW);
  digitalWrite(term_vcc, HIGH);
  delay(100);
  sensors.requestTemperatures();
  delay(500);
  sensors.requestTemperatures();
  t = float(sensors.getTempCByIndex(0));
  t = t + 0.0;
  digitalWrite(term_vcc, LOW);
  delay(1400);
}
```

jockethebastard, 2024-03-18:
> > > @louisdcmr https://github.com/purpln/homekit/tree/master/examples/sensor_temperature
> > 
> > 
> > Would you mind sharing this again? I'm having such a hard time configuring the temperature values. The sensor shows up in HomeKit, no problem there, but the values changes randomly
> 
> * [HomeKit](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266)
> * [DallasTemperature](https://github.com/milesburton/Arduino-Temperature-Control-Library)
> * [OneWire](https://github.com/PaulStoffregen/OneWire)
> 
> ### temperature.c
> ```c
> #include <Arduino.h>
> #include <homekit/types.h>
> #include <homekit/homekit.h>
> #include <homekit/characteristics.h>
> #include <stdio.h>
> #include <port.h>
> 
> #define ACCESSORY_NAME  ("temperature")
> #define ACCESSORY_SN  ("41745062")
> #define ACCESSORY_MANUFACTURER ("homekit")
> #define ACCESSORY_MODEL  ("esp8285")
> 
> homekit_characteristic_t name = HOMEKIT_CHARACTERISTIC_(NAME, ACCESSORY_NAME);
> homekit_characteristic_t serial_number = HOMEKIT_CHARACTERISTIC_(SERIAL_NUMBER, ACCESSORY_SN);
> homekit_characteristic_t temperature = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0);
> float temperature_value;
> extern float t;
> 
> void temp_update() {
>   temperature_value = t;
>   temperature.value.float_value = temperature_value;
>   *temperature.min_value = -200;
>   *temperature.max_value = 200;
>   homekit_characteristic_notify(&temperature, HOMEKIT_FLOAT(temperature_value));
>   delay(1000);
> }
> 
> void temperature_sensor_identify(homekit_value_t _value) {
>   printf("Temperature sensor identify\n");
> }
> 
> homekit_accessory_t *accessories[] = {
>   HOMEKIT_ACCESSORY(.id = 1, .category = homekit_accessory_category_thermostat, .services=(homekit_service_t*[]){
>     HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
>       &name,
>       HOMEKIT_CHARACTERISTIC(MANUFACTURER, ACCESSORY_MANUFACTURER),
>       &serial_number,
>       HOMEKIT_CHARACTERISTIC(MODEL, ACCESSORY_MODEL),
>       HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
>       HOMEKIT_CHARACTERISTIC(IDENTIFY, temperature_sensor_identify),
>       NULL
>     }),
>     HOMEKIT_SERVICE(TEMPERATURE_SENSOR, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
>       HOMEKIT_CHARACTERISTIC(NAME, "temperature sensor"),
>       &temperature,
>       NULL
>     }),
>     NULL
>   }),
>   NULL
> };
> 
> homekit_server_config_t config = {
>   .accessories = accessories,
>   .password = "111-11-111",
> };
> ```
> 
> ### sensor_temperature.ino
[... 71 lines cut ...]
>   arduino_homekit_loop();
>   static uint32_t next_heap_millis = 0;
>   uint32_t time = millis();
>   if (time > next_heap_millis) {
>     SIMPLE_INFO("heap: %d, sockets: %d",
>         ESP.getFreeHeap(), arduino_homekit_connected_clients_count());
>     next_heap_millis = time + 5000;
>   }
> }
> 
> void temperatures() {
>   digitalWrite(term_gnd, LOW);
>   digitalWrite(term_vcc, HIGH);
>   delay(100);
>   sensors.requestTemperatures();
>   delay(500);
>   sensors.requestTemperatures();
>   t = float(sensors.getTempCByIndex(0));
>   t = t + 0.0;
>   digitalWrite(term_vcc, LOW);
>   delay(1400);
> }
> ```

Thanks for replying, appreciate it! Unfortunately I can’t get it working either. It showed up after several tries, but showing -127 degrees and no response. I didn’t know if the wifi_info.h should be included or not, but only after adding that file it showed up, but not working. Don’t know what to make of this, since I have no problem with the switch sketch, and getting the thermometer to show in HK is no problem. I’ve also been able to see the thermometer in work with at different sketch (non HomeKit). With no programming experience whatsoever I’ve spent hours and hours reading and comparing code, hoping to find something, but it’s always a dead end. Thanks again for sharing the code, I’ll keep on searching