Mixiaoxiao/Arduino-HomeKit-ESP8266 #16: ESP32 fatal error: osapi.h: No such file or directory compilation terminated.

sto-chastic, 2020-05-16:
I get this error when compiling the example for ESP32:

> Arduino/libraries/Arduino-HomeKit-ESP8266-master/src/user_settings.h:15:19: fatal error: osapi.h: No such file or directory
> compilation terminated.

sto-chastic, 2020-05-16:
This was being caused as it was using the ESP8266 src. Maybe this is not the cleanest solution, but I replaced the src in my installed Arduino libraries with the src for the ESP32. <Ugly solution but worked

Mixiaoxiao, 2020-05-16:
You should use the library in "ESP32 HomeKit" folder (download it from github and install it manually).
The libraries for ESP8266 and ESP32 are different since these two chips are of different SDK(s).

closed 2020-05-18 by Mixiaoxiao