Mixiaoxiao/Arduino-HomeKit-ESP8266 #17: Example03 and Example04 not Pairing

spxnr, 2020-05-18:
Hi Mixiaoxiao,

You have done amazing work with this library. First and foremost, thank you.
Example01 and Example02 sketches work great when flashed on my ESP8266.

I am having difficulty pairing with my ESP8266 when flashed with Example03 and Example04.

Here are the serial logs:
**Example03**
NOTE: It appears that it attempts to pair twice.

`WiFi connecting...
.......................................
WiFi connected, IP: 192.168.0.130
>>> [   4799] HomeKit: Starting server
>>> [   4805] HomeKit: Using existing accessory ID: FF:CF:78:7A:AA:A4
>>> [   4811] HomeKit: Preiniting pairing context
>>> [   4816] HomeKit: Using user-specified password: 111-11-111
>>> [   4837] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  10904] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [  14023] HomeKit: Preinit pairing context success
>>> [  14028] HomeKit: Configuring MDNS
>>> [  14033] HomeKit: MDNS.begin: Stateless Programmable Switch, IP: 192.168.0.130
>>> [  14042] HomeKit: Init server over
Free heap: 41632, HomeKit clients: 0
Free heap: 42200, HomeKit clients: 0
Free heap: 41848, HomeKit clients: 0
Free heap: 42168, HomeKit clients: 0
Free heap: 42200, HomeKit clients: 0
Free heap: 42168, HomeKit clients: 0
Free heap: 41848, HomeKit clients: 0
>>> [  46647] HomeKit: Got new client: local 192.168.0.130:5556, remote 192.168.0.228:49758
>>> [  46657] HomeKit: [Client 1073682612] Pair Setup Step 1/3
Free heap: 40584, HomeKit clients: 1
>>> [  50178] HomeKit: [Client 1073682612] Pair Setup Step 2/3
>>> [  50190] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  58947] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
Free heap: 40224, HomeKit clients: 1
>>> [  63575] HomeKit: [Client 1073682612] Pair Setup Step 3/3
>>> [  63605] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  64357] HomeKit: Added pairing with 0C73D7AF-93CA-425E-A190-9DA038331F4A
>>> [  64433] HomeKit: Free saved_preinit_pairing_context
>>> [  64438] HomeKit: [Client 1073682612] Successfully paired
>>> [  64475] HomeKit: [Client 1073682612] Disconnected!
>>> [  64480] HomeKit: [Client 1073682612] Closing client connection
>>> [  64617] HomeKit: Got new client: local 192.168.0.130:5556, remote 192.168.0.228:49759
>>> [  64638] HomeKit: [Client 1073682940] Pair Verify Step 1/2
>>> [  64962] HomeKit: Free heap: 42344
>>> [  65089] HomeKit: [Client 1073682940] Pair Verify Step 2/2
>>> [  65095] HomeKit: [Client 1073682940] Found pairing with 0C73D7AF-93CA-425E-A190-9DA038331F4A
>>> [  65119] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  65883] HomeKit: [Client 1073682940] Verification successful, secure session established
>>> [  65891] HomeKit: Free heap: 42456
>>> [  65958] HomeKit: [Client 1073682940] Get Accessories
>>> [  66035] HomeKit: [Client 1073682940] Remove Pairing
>>> [  66040] HomeKit: Removed pairing with 0C73D7AF-93CA-425E-A190-9DA038331F4A
>>> [  66048] HomeKit: Last admin pairing was removed, enabling pair setup
>>> [  66069] HomeKit: Preiniting pairing context
>>> [  66073] HomeKit: Using user-specified password: 111-11-111
>>> [  66095] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  72119] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [  75205] HomeKit: Preinit pairing context success
>>> [  75212] HomeKit: [Client 1073682940] Disconnected!
>>> [  75218] HomeKit: [Client 1073682940] Closing client connection
Free heap: 41504, HomeKit clients: 0
Free heap: 41944, HomeKit clients: 0
Free heap: 41624, HomeKit clients: 0
Free heap: 41624, HomeKit clients: 0`

**Example04**
NOTE: Device is discoverable as "Multiple Sensors", and lets me type the 111-11-111 pairing code, but it doesn't pair successfully. It appears that the pairing process doesn't start according to this log...

`WiFi connecting...
.................................................
WiFi connected, IP: 192.168.0.130
>>> [   5946] HomeKit: Starting server
>>> [   5952] HomeKit: Formatting HomeKit storage at 0xfb000
>>> [   5958] HomeKit: Generated new accessory ID: AE:9B:49:9C:E2:28
>>> [   6019] HomeKit: Generated new accessory key
>>> [   6024] HomeKit: Preiniting pairing context
>>> [   6029] HomeKit: Using user-specified password: 111-11-112
>>> [   6050] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  12073] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [  15191] HomeKit: Preinit pairing context success
>>> [  15196] HomeKit: Configuring MDNS
>>> [  15202] HomeKit: MDNS.begin: Multiple Sensors, IP: 192.168.0.130
>>> [  15209] HomeKit: Init server over
t 11.0, h 57.0, l 3446.0, c 1, m 0, o 1
Free heap: 37912, HomeKit clients: 0
Free heap: 38288, HomeKit clients: 0
t 19.0, h 39.0, l 9947.0, c 0, m 1, o 1
Free heap: 38128, HomeKit clients: 0
Free heap: 38448, HomeKit clients: 0
t 24.0, h 49.0, l 8718.0, c 1, m 0, o 1
Free heap: 38128, HomeKit clients: 0
Free heap: 38448, HomeKit clients: 0
t 11.0, h 30.0, l 6588.0, c 0, m 1, o 1
Free heap: 38448, HomeKit clients: 0
Free heap: 38456, HomeKit clients: 0
t 28.0, h 59.0, l 5009.0, c 0, m 1, o 0
Free heap: 38448, HomeKit clients: 0
Free heap: 38128, HomeKit clients: 0
t 16.0, h 55.0, l 7712.0, c 1, m 0, o 1
Free heap: 38448, HomeKit clients: 0
Free heap: 38448, HomeKit clients: 0
t 20.0, h 34.0, l 7964.0, c 0, m 0, o 0
Free heap: 38272, HomeKit clients: 0
Free heap: 38272, HomeKit clients: 0
t 18.0, h 33.0, l 2112.0, c 1, m 1, o 0
Free heap: 38448, HomeKit clients: 0
Free heap: 37848, HomeKit clients: 0`

Mixiaoxiao, 2020-05-18:
**Example03**
The behavior that "[ 65883] Verification successful" -> "[ 65958] Get Accessories" -> "[ 66035] Remove Pairing", is generally caused by the illegal accessory definition/value for Home app. What is your iOS version? I am using 13.4.1 and this example works fine.

**Example04**
Just try again. That is true that the pairing process did not start. The TCP socket connection is not received (accepted) on ESP. Maybe something wrong in WiFi connection or your network. 
You should start to add accessory 2~3 seconds after waiting for “[ 15209] Init server over”. ESP8266 is not based on RTOS and the WiFi connection is not working while Preiniting pairing context (around 9 seconds).

Mixiaoxiao, 2020-05-25:
Will be closed in a week since there is no following response.

closed 2020-06-01 by Mixiaoxiao

vniehues, 2021-06-22:
I know this was closed a long time ago but im facing the exact same problems now using iOS 14 and iOS 15 B1