Mixiaoxiao/Arduino-HomeKit-ESP8266 #18: in_use seems to throw errors despite being a requirement for a valve

Lethegrin, 2020-05-20:
Trying to create a valve / sprinkler, however can't seem to get "in_use" working. Could you provide an example that makes use of characteristic type valve?

Mixiaoxiao, 2020-05-20:
Refer to the Apple's HAP section 8.43. The 'Valve' seems to be a little bit complicated service.

Lethegrin, 2020-05-20:
I was able to get it to compile by adding this.
 `  homekit_characteristic_t cha_in_use = HOMEKIT_CHARACTERISTIC_(IN_USE, false);`

 However the device continuously says "updating". Perhaps in_use is an int8?

Mixiaoxiao, 2020-05-20:
uint8. Defined in Apple's HAP section 9.48.

Lethegrin, 2020-05-20:
updated all values to uint8. Looked at HAP 8.43 and it describes the only values needed are Active, In Use, and Valve Type if the valve type is anything other than 1 "Irrigation System". Still getting a valve that is stuck in "updating..."

nunogbento, 2024-05-21:
> updated all values to uint8. Looked at HAP 8.43 and it describes the only values needed are Active, In Use, and Valve Type if the valve type is anything other than 1 "Irrigation System". Still getting a valve that is stuck in "updating..."

I'm getting the same behavior. HK client keeps disconnecting when reading characteristics. In use characteristic seems to be not updated most of the times.

19:58:09.885 -> >>> [ 826949] HomeKit: [Client 1073685516] Get Accessories
19:58:09.978 -> >>> [ 827016] HomeKit: [Client 1073685516] Get Characteristics
19:58:09.978 -> >>> [ 827029] HomeKit: [Client 1073685516] Get Characteristics
19:58:20.130 -> >>> [ 837191] HomeKit: [Client 1073685516] Disconnected!
19:58:20.130 -> >>> [ 837196] HomeKit: [Client 1073685516] Closing client connection