Mixiaoxiao/Arduino-HomeKit-ESP8266 #19: Discussion about future work about HomeKit on ESP8266 and ESP32

Mixiaoxiao, 2020-05-20:
I notice that Espressif has published its official HomeKit library in this month, available on Espressif's GitHub repository [esp-apple-homekit-adk](https://github.com/espressif/esp-apple-homekit-adk).
This official HomeKit library based on ESP-IDF is for ESP32 and ESP32-S2 only.
Maybe I will do something to make this official HomeKit library work in Arduino environment. 
And I would no longer maintain the HomeKit library for ESP32 based on the unofficial project [esp-homekit](https://github.com/maximkulkin/esp-homekit).

a914622719, 2020-05-21:
期代！

vvrec, 2020-05-22:
Thank you for your work! It would be cool if it were possible to program and flash ESP32 in Arduino IDE. Along with the fixed bug of the cliff of the live broadcast. Thank! 🙏

milanek, 2020-06-24:
Thank you for your work on ESP8266 - it works nicely and pairing works in 9/10 cases @160 MHz.
I would appreciate official HomeKit for ESP32 support in Arduino IDE as I would like to have more horse power.
Meanwhile I will try to use your "deprecated" library after fixing installation issue on Mac.