Mixiaoxiao/Arduino-HomeKit-ESP8266 #21: Unable to use this library for ESP32

jsve, 2020-05-23:
This looks like an awesome project, and would be exactly what I need. However; I'm having some issues getting this to work on my ESP32 board. The issue is probably a case of me not understanding the structure in your repo, and/or not understanding how to properly use an external library.

Here is what i have done: 
 1. download the repo as a zip
 2. added the zip as a library in the arduino IDE
 2. replace the src folder (`Sketches/MyProject/libraries/Arduino-HomeKit-ESP8266-master/src`) with the one in the `ESP32 HomeKit` folder.
3. copy the example files from the `ESP32 HomeKit` into MyProject
4. try to compile

Since my hypothesis is that I'm setting this up wrong, I won't bother you with the weird error messages I'm getting. Could you give me some pointers as to how to get this running on my ESP32? I'd be happy to help you describe the workflow in the readme, once I have gotten it running!

Mixiaoxiao, 2020-05-24:
What does your “unable to use” means? Not compile? Build the example but not work?

jsve, 2020-05-24:
should probably have added nr. 6 - Does not compile. I'm getting a bunch of errors when trying to compile.

Mixiaoxiao, 2020-05-24:
I am sure you did something wrong. But there is no detail about what you did.

jsve, 2020-05-25:
Which of the 5 (now 6) steps is it I need to elaborate on?

Here are the errors i get when trying to compile:

```
WARNING: library Arduino-HomeKit-ESP8266-master claims to run on esp8266 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from /somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:3:0:
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp: In function 'void homekit_server_on_pair_verify(client_context_t*, const byte*, size_t)':
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/homekit_debug.h:40:43: warning: 'uint32_t system_get_free_heap_size()' is deprecated [-Wdeprecated-declarations]
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                                           ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/esp_xpgm.h:28:56: note: in definition of macro 'XPGM_PRINTF'
 #define XPGM_PRINTF(fmt, ...)   printf_P(PSTR(fmt) , ##__VA_ARGS__);
                                                        ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/homekit_debug.h:40:21: note: in expansion of macro 'INFO'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                     ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:1774:2: note: in expansion of macro 'INFO_HEAP'
  INFO_HEAP();
  ^
In file included from /somepath/myproject/hardware/espressif/esp32/tools/sdk/include/freertos/freertos/portable.h:128:0,
                 from /somepath/myproject/hardware/espressif/esp32/tools/sdk/include/freertos/freertos/FreeRTOS.h:105,
                 from /somepath/myproject/hardware/espressif/esp32/cores/esp32/Arduino.h:32,
                 from /somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:2:
/somepath/myproject/hardware/espressif/esp32/tools/sdk/include/esp32/esp_system.h:141:10: note: declared here
 uint32_t system_get_free_heap_size(void)  __attribute__ ((deprecated));
          ^
In file included from /somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:3:0:
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/homekit_debug.h:40:43: warning: 'uint32_t system_get_free_heap_size()' is deprecated [-Wdeprecated-declarations]
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                                           ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/esp_xpgm.h:28:56: note: in definition of macro 'XPGM_PRINTF'
 #define XPGM_PRINTF(fmt, ...)   printf_P(PSTR(fmt) , ##__VA_ARGS__);
                                                        ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/homekit_debug.h:40:21: note: in expansion of macro 'INFO'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                     ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:1774:2: note: in expansion of macro 'INFO_HEAP'
  INFO_HEAP();
  ^
In file included from /somepath/myproject/hardware/espressif/esp32/tools/sdk/include/freertos/freertos/portable.h:128:0,
                 from /somepath/myproject/hardware/espressif/esp32/tools/sdk/include/freertos/freertos/FreeRTOS.h:105,
                 from /somepath/myproject/hardware/espressif/esp32/cores/esp32/Arduino.h:32,
                 from /somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:2:
/somepath/myproject/hardware/espressif/esp32/tools/sdk/include/esp32/esp_system.h:141:10: note: declared here
 uint32_t system_get_free_heap_size(void)  __attribute__ ((deprecated));
          ^
In file included from /somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:3:0:
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/homekit_debug.h:40:69: warning: 'uint32_t system_get_free_heap_size()' is deprecated [-Wdeprecated-declarations]
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                                                                     ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/esp_xpgm.h:28:56: note: in definition of macro 'XPGM_PRINTF'
 #define XPGM_PRINTF(fmt, ...)   printf_P(PSTR(fmt) , ##__VA_ARGS__);
                                                        ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/homekit_debug.h:40:21: note: in expansion of macro 'INFO'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                     ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:1774:2: note: in expansion of macro 'INFO_HEAP'
  INFO_HEAP();
  ^
In file included from /somepath/myproject/hardware/espressif/esp32/tools/sdk/include/freertos/freertos/portable.h:128:0,
                 from /somepath/myproject/hardware/espressif/esp32/tools/sdk/include/freertos/freertos/FreeRTOS.h:105,
                 from /somepath/myproject/hardware/espressif/esp32/cores/esp32/Arduino.h:32,
                 from /somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/arduino_homekit_server_esp32.cpp:2:
[... cut ...]
 void system_restart(void) __attribute__ ((deprecated, noreturn));
      ^
/somepath/myproject/libraries/Arduino-HomeKit-ESP8266-master/src/wolfcrypt/src/evp.c:26:10: warning: #warning evp.c does not need to be compiled separately from ssl.c [-Wcpp]
         #warning evp.c does not need to be compiled separately from ssl.c
          ^
Sketch uses 900004 bytes (68%) of program storage space. Maximum is 1310720 bytes.
Global variables use 41984 bytes (12%) of dynamic memory, leaving 285696 bytes for local variables. Maximum is 327680 bytes.
```

Mixiaoxiao, 2020-05-25:
Tell me, which line shows an error?

Mixiaoxiao, 2020-06-23:
Check out the new repository for ESP32 [Arduino-HomeKit-ESP32](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP32).

The log you shown above has only `WARNING` (does not matter to compile), no `ERROR`.

closed 2020-07-04 by Mixiaoxiao

stanjp1, 2022-12-15:
Hello , 

I have the same issu when using the exemple with the same errors 👍 

Arduino : 1.8.19 (Mac OS X), Carte : "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

In file included from /Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/arduino_homekit_server_esp32.cpp:3:
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/arduino_homekit_server_esp32.cpp: In function 'void homekit_server_on_pair_verify(client_context_t*, const byte*, size_t)':
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/homekit_debug.h:40:43: error: 'system_get_free_heap_size' was not declared in this scope
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/esp_xpgm.h:28:56: note: in definition of macro 'XPGM_PRINTF'
 #define XPGM_PRINTF(fmt, ...)   printf_P(PSTR(fmt) , ##__VA_ARGS__);
                                                        ^~~~~~~~~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/homekit_debug.h:40:21: note: in expansion of macro 'INFO'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                     ^~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/arduino_homekit_server_esp32.cpp:1774:2: note: in expansion of macro 'INFO_HEAP'
  INFO_HEAP();
  ^~~~~~~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/homekit_debug.h:40:43: note: suggested alternative: 'esp_get_free_heap_size'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/esp_xpgm.h:28:56: note: in definition of macro 'XPGM_PRINTF'
 #define XPGM_PRINTF(fmt, ...)   printf_P(PSTR(fmt) , ##__VA_ARGS__);
                                                        ^~~~~~~~~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/homekit_debug.h:40:21: note: in expansion of macro 'INFO'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                     ^~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/arduino_homekit_server_esp32.cpp:1774:2: note: in expansion of macro 'INFO_HEAP'
  INFO_HEAP();
  ^~~~~~~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/arduino_homekit_server_esp32.cpp: In function 'void homekit_server_init(homekit_server_config_t*)':
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/arduino_homekit_server_esp32.cpp:3439:4: error: 'system_restart' was not declared in this scope
    system_restart();
    ^~~~~~~~~~~~~~
/Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32/src/arduino_homekit_server_esp32.cpp:3439:4: note: suggested alternative: 'system_event_t'
    system_restart();
    ^~~~~~~~~~~~~~
    system_event_t
Plusieurs bibliothèque trouvées pour "WiFi.h"
Utilisé : /Users/thomasleguillou/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5/libraries/WiFi
Non utilisé : /private/var/folders/fs/6dh8w0s122qdmtmwtwm9jtp00000gn/T/AppTranslocation/DE2F1324-8BF8-4CD9-B97C-5A1AADF040E7/d/Arduino.app/Contents/Java/libraries/WiFi
Plusieurs bibliothèque trouvées pour "arduino_homekit_server.h"
Utilisé : /Users/thomasleguillou/Documents/Arduino/libraries/Arduino_HomeKit_ESP32
Non utilisé : /Users/thomasleguillou/Documents/Arduino/libraries/Arduino-HomeKit-ESP32-master
exit status 1
Erreur de compilation pour la carte ESP32 Dev Module

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

DeveloperSpawn, 2023-04-17:
I have something close, most of my erros refer to arduino_homekit_server_esp32.cpp file. I think it is related to libraries versions my error list looks like this: 

```
...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src\arduino_homekit_server_esp32.cpp:3:

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src\arduino_homekit_server_esp32.cpp: In function 'void homekit_server_on_pair_verify(client_context_t*, const byte*, size_t)':

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src/homekit_debug.h:40:43: error: 'system_get_free_heap_size' was not declared in this scope
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src/esp_xpgm.h:28:56: note: in definition of macro 'XPGM_PRINTF'
 #define XPGM_PRINTF(fmt, ...)   printf_P(PSTR(fmt) , ##__VA_ARGS__);
                                                        ^~~~~~~~~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src/homekit_debug.h:40:21: note: in expansion of macro 'INFO'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                     ^~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src\arduino_homekit_server_esp32.cpp:1774:2: note: in expansion of macro 'INFO_HEAP'
  INFO_HEAP();
  ^~~~~~~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src/homekit_debug.h:40:43: note: suggested alternative: 'esp_get_free_heap_size'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                                           ^~~~~~~~~~~~~~~~~~~~~~~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src/esp_xpgm.h:28:56: note: in definition of macro 'XPGM_PRINTF'
 #define XPGM_PRINTF(fmt, ...)   printf_P(PSTR(fmt) , ##__VA_ARGS__);
                                                        ^~~~~~~~~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src/homekit_debug.h:40:21: note: in expansion of macro 'INFO'
 #define INFO_HEAP() INFO("Free heap: %u", system_get_free_heap_size());
                     ^~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src\arduino_homekit_server_esp32.cpp:1774:2: note: in expansion of macro 'INFO_HEAP'
  INFO_HEAP();
  ^~~~~~~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src\arduino_homekit_server_esp32.cpp: In function 'void homekit_server_init(homekit_server_config_t*)':

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src\arduino_homekit_server_esp32.cpp:3439:4: error: 'system_restart' was not declared in this scope
    system_restart();
    ^~~~~~~~~~~~~~

...\Arduino\libraries\Arduino-HomeKit-ESP32-master\src\arduino_homekit_server_esp32.cpp:3439:4: note: suggested alternative: 'system_event_t'
    system_restart();
    ^~~~~~~~~~~~~~
    system_event_t

Using library WiFi at version 2.0.0 in folder: 
...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\WiFi 
Using library HomeKit-ESP32 at version 1.0.0 in folder: 
...\Arduino\libraries\Arduino-HomeKit-ESP32-master 
Using library Ticker at version 2.0.0 in folder: ...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\Ticker 
Using library ESPmDNS at version 2.0.0 in folder: ...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\libraries\ESPmDNS 

```

Compiling in Arduino IDE 2.0.3
Using board 'node32s' from platform in folder: C:\Users\Sensei\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6
Using core 'esp32' from platform in folder: C:\Users\Sensei\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6

I used you sketch before and succsesfully improved it with ESPConnect. How I did it before:  I played with different versions of libraries and it worked, but now I got this errors even with your reference