Mixiaoxiao/Arduino-HomeKit-ESP8266 #22: I get an error while compiling

ethem4161, 2020-05-24:
where do you think there is an error?
[image]
[image]
[image]
[image]
 I'm a little new at this, sorry :/
If you can shoot a short video while compiling, I can see where I did wrong. Thanks
Good Works.

Mixiaoxiao, 2020-05-24:
Hi, guy, it is not my duty to teach you how to code.

`#include <ESP8266WiFi.h>`

`const char *ssid = "your-ssid";`
`const char *password = "your-password";`

ethem4161, 2020-05-25:
"Hi, Thank you for helping."
"I'm new to this software. sorry ;/" 
"I have encountered such a problem now; "

Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "Generic ESP8266 Module, 160 MHz, Flash, Legacy (new can return nullptr), Basic SSL ciphers (lower ROM use), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+119 (191122), v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

c:/users/dell/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\sketch_may24a.ino.cpp.o:(.text.setup+0xc): undefined reference to `config'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "WiFiServer.h"
 Used: C:\Users\DELL\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.3\libraries\ESP8266WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

"I also tried linux operating system. Same mistake"

[image]

Good Works Guy.

Mixiaoxiao, 2020-05-25:
Again, it is not my duty to teach you how to code.
I will not give any feedback in this issue, since it is really not an issue about this project.

> undefined reference to `config'

ethem4161, 2020-05-26:
Thank you for helping.
I wish you good work.

wahidsa, 2020-06-25:
ethtem,
You are copy/paste script from the github page, NO!! this way it will not work.
Download the whole project as a .zip, open it through clicking the file name, this way it will load extra tabs, and everything is super.
Mixiaoxiao did a really hard and good job, but this is really not an issue.