Mixiaoxiao/Arduino-HomeKit-ESP8266 #24: Question about delay() in the main loop...

vvrec, 2020-05-27:
If I understand correctly - delay in main loop is needed for the correct work of RTOS.
Can I use Ticker, or millisecond around all my functions in the loop instead of delay () to let RTOS to work 5, 10 ms, and use other timers in parallel with a interval longer then 5, 10 ms? Or are they anyway mix with each other and there may be problems?

Mixiaoxiao, 2020-05-27:
Not true. This seems like not an issus about this library, but ESP8266 itself.
ESP8266 Arduino Core is not based on RTOS.
Refer to the section "Interrupts" and "Timing and delays" in [https://arduino-esp8266.readthedocs.io/en/2.7.1/reference.html](https://arduino-esp8266.readthedocs.io/en/2.7.1/reference.html).

vvrec, 2020-05-27:
Thanks for the link! Already read it. But I see that there is a similar delay(5) in the example for esp32... Which has a coprocessor ...

Mixiaoxiao, 2020-05-27:
That (delay in ESP32) does not matter. The delay() is vTaskDelay() on ESP32 indeed, which is different from ESP8266. ESP32 Arduino is based on RTOS and vTaskDelay is a RTOS api. Writing the delay is to make the loop-task "not always consume the CPU".

vvrec, 2020-05-27:
Thanks for the explanation!

closed 2020-05-27 by vvrec