Mixiaoxiao/Arduino-HomeKit-ESP8266 #28: Apple 家庭搜不到设备是什么原因呢

yangkai6667, 2020-06-14:
第一次烧写后可以正常搜到设备，第二次烧写完成后就再也搜不到设备了

yangkai6667, 2020-06-14:
配对的时候有时候可以搜到设备ESP8266_LED 但添加配件的时候无法成功，显示无法添加配件

Mixiaoxiao, 2020-06-14:
贴出来你的日志

yangkai6667, 2020-06-14:
> 贴出来你的日志

[image]

yangkai6667, 2020-06-14:
> 贴出来你的日志

您好，串口中打印的信息一直是这样，希望您能帮我指导一下，感激不尽！！

yangkai6667, 2020-06-14:
> 贴出来你的日志

[image]
这是我的配置，基本都是参考您给的推荐配置设置的

yangkai6667, 2020-06-14:
> 贴出来你的日志

[image]
我重新连接了一次，这次串口打印的日志是这样的

Mixiaoxiao, 2020-06-14:
贴一下你的accessory定义

yangkai6667, 2020-06-14:
> 贴一下你的accessory定义

[image]
我用的是1.1.0的版本实例simplest_led, accessory部分没有改动

Mixiaoxiao, 2020-06-14:
simplest_led示例项目其他地方有改动吗？
出错的地方在 [ 161838]那行日志，(注：!!! 开头的日志代表这行是error)，这行出错信息是指：Socket需要传输525字节数据，但实际只传了411字节。
可能是网络方面的原因，你可以换个开发板或者换个路由器或者改善一下WIFI信号试试。
你日志里面error的本身不属于本library的bug。

另外关于你的issue的标题提出的问题，如果你的iOS与某个配件(多次)配对失败，iOS就认为这个配件有问题，就不会在搜索列表里面显示这个配件了，这是我个人的经验，没有官方说明。可行的解决方法：给你的配件换个名字，或者重启一下你的iPhone。

yangkai6667, 2020-06-14:
> simplest_led示例项目其他地方有改动吗？
> 出错的地方在 [ 161838]那行日志，(注：!!! 开头的日志代表这行是error)，这行出错信息是指：Socket需要传输525字节数据，但实际只传了411字节。
> 可能是网络方面的原因，你可以换个开发板或者换个路由器或者改善一下WIFI信号试试。
> 你日志里面error的本身不属于本library的bug。 [...]

非常感谢，因为我用的是电脑热点，可能信号比较弱，我多连接几次试试，再次感谢您的帮助

Mixiaoxiao, 2020-06-23:
可以尝试一下如下改动，可能会解决你日志中 [ 161838]那行的问题：
在源码`arduino_homekit_server.cpp`文件中，用下面的`write`函数替换原有的`write`（约在520行左右的位置）

```C   
void write(client_context_t *context, byte *data, int data_size) {
	if ((!context) || (!context->socket) || (!context->socket->connected())) {
		CLIENT_ERROR(context, "The socket is null! (or is closed)");
		return;
	}
	if (context->error_write) {
		CLIENT_ERROR(context, "Abort write data since error_write.");
		return;
	}
	int bytesWritten = 0;
	const byte *p = data;
	while (bytesWritten < data_size) {
		int written;
		int towrite = std::min((int) (data_size - bytesWritten), (int) HOMEKIT_JSONBUFFER_SIZE);
		written = context->socket->write(p + bytesWritten, towrite);
		if (written <= 0) {
			context->error_write = true;
			context->disconnect = true;
			homekit_server_close_client(context->server, context);
			return;
		}
		if(written != towrite){
			CLIENT_INFO(context, "socket.write, towrite=%d, written=%d", towrite, written);
		}
		bytesWritten += written;
		delay(10);
	}
}
```

ctudoudou, 2020-09-12:
Erase full flash for the ESP and then re-flash the ESP, and solved my problem.

Mixiaoxiao, 2020-10-02:
建议不要使用手机/电脑热点或者质量较差的WiFi网络，在存在较大网络丢包率的情况下配对数据容易出错。

closed 2020-10-02 by Mixiaoxiao