Mixiaoxiao/Arduino-HomeKit-ESP8266 #29: No yield()

vaddieg, 2020-06-21:
Thank you very much for the library. 
Have you tried to put yield() after every SrpHashUpdate() call inside the most painful functions like wolfcrypt::wc_SrpComputeKey() ? 
It should solve watchdog issues and improve connection stability

Mixiaoxiao, 2020-06-23:
This is a good issue. 
First, the answer for your question is NO.

I have struggled with the wtd(watchdog) for a long time before I made this library work.
Actually, the most painful function is the `mp_exptmod` in `integer.c`.

- Pair Setup 1/3 (= Preinit)  total ~9s
  (1) `crypto_srp_init`    ~6s
      includes ① `mp_exptmod`(winsize=6)      ~6s
  (2) `crypto_srp_get_public_key`  ~3s
      includes ② `mp_exptmod`(winsize=5)   ~3s   

- Pair Setup 2/3  total ~12s
  (3) `crypto_srp_compute_key` (2 calls of mp_exptmod)  ~12s
      includes ③ `mp_exptmod` (winsize=6)      ~8s
      and ④ `mp_exptmod`(winsize=5)      ~4s
Note: running on CPU@160Mhz.

The watchdogs (both hardware and software watchdog ) are disabled in these (1)(2)(3) functions. And as you can see, the 99% of total time is cost by `mp_exptmod`.

The hardware watchdog of ESP8266 is ~8.2s, and the software watchdog is ~3.2. (Refer to [esp8266/arduino/watchdogs_en](https://www.sigmdel.ca/michel/program/esp8266/arduino/watchdogs_en.html#ESP8266_WDT_TIMEOUT))

I know it is not good to disable the watchdogs for running long time functions since it will lead to a unstable WiFi connection. But in my tests, it did not show any visible bad influence to WiFi connection since these long time functions are not that too long. On the other hand, I am also finding a better way to handle this issue. In my opinion, it is not easy and good to insert `yield()` or `delay()` in `mp_exptmod`.

Anyway, this issue is caused by (the low performance && the ESP8266_NONOS_SDK). Note that the `mp_exptmod` on ESP32 is mush faster than ESP8266 even without hardware acceleration.

BTW, the new ESP32-S2 with the following features may be a good replacement of ESP8266 to implement HomeKit projects:
1. New Xtensa single-core 32-bit LX7 @240MHz (ESP32 is dual LX6 @240MHz, ESP8266 is L106 @160MHz, maybe ESP32-S2 is even faster than ESP32)
1. WiFi only, no Bluetooth, lower price than ESP32 (maybe same as ESP8266)
1. Crypto co-processors (will have a hardware accelerated `mp_exptmod`)

vaddieg, 2020-06-23:
ESP32-S2 appears to be a perfect choice due to low-power idling Wi-Fi. Unfortunately, boards aren't yet widely available and official espressif/esp-apple-homekit-adk does not integrate to arduino IDE, that's why I like this project so much.
Hardcoding yield/delay into integer.c or wolfcrypt isn't a good idea, but injecting optional timeout_callback function (NULL by default, {yield()} when called from time-heavy context) might be the way to go

Mixiaoxiao, 2020-06-24:
what does your `injecting optional timeout_callback function` mean?

vaddieg, 2020-06-24:
regular function pointer, pointing to empty void func by default