Mixiaoxiao/Arduino-HomeKit-ESP8266 #31: HOMEKIT_MAX_CLIENTS - why 8?

jmbwell, 2020-06-24:
Why is the default HOMEKIT_MAX_CLIENTS set to 8? It would be helpful to increase this, in homes with lots of Apple devices.

Mixiaoxiao, 2020-06-24:
This value needs to be considered. 
The original value in esp-homekit library based on esp-open-rtos is 16. But in Arduino, there is an internal limitation of max TCP connection, which is 5 default. Refer to MEMP_NUM_TCP_PCB(5) and espconn_tcp_set_max_con. So setting HOMEKIT_MAX_CLIENTS to 8 maybe useless here, and I don't have so many iOS devices to test.
Anther thing is that one tcp connection requests aroud 2k heap memory in my test and free heap of esp8266 is around 40k.

jmbwell, 2020-07-02:
It looks like MEMP_NUM_TCP_PCB is set to 5 in lwip2, but in lwip, which seems to be what is being used, it's doing this to read it from a register (which I don't understand):

```
/**
 * MEMP_NUM_TCP_PCB: the number of simulatenously active TCP connections.
 * (requires the LWIP_TCP option)
 */
#ifndef MEMP_NUM_TCP_PCB
#define MEMP_NUM_TCP_PCB                (*(volatile uint32*)0x600011FC)
#endif
```
Also, if I simply increase HOMEKIT_MAX_CLIENTS in arduino_homekit_server.cpp to, say, 15, I'm able to get more clients connected (I've managed up to 12). So we seem to be able to get more connections than lwip is set to allow (which I don't understand). If lwip is zeroing out and re-using the older connections, then HomeKit clients handle that rather gracefully… client devices don't seem to care.

It occurs to me that using espconn_tcp_get_max_con to retrieve the maximum value from the sdk, rather than setting it manually, might be an approach to consider. All it appears to do is just return MEMP_NUM_TCP_PCB after all.

On the other hand, if it works just to set a higher number of max clients and let HomeKit recover from dropped connections on the client side, maybe that's just fine.

Increasing connections does reduce available heap.

In any case, there is a lot here I don't understand!

Mixiaoxiao, 2020-07-04:
I wrote this number 8 at random when I wrote this library. Anyway it depends on your real project and what you really need, and there is no best value to be written in a library. You can increase it or reduce it by modifying the source code. Also I do think ESP8266 is not so powerful to hold so many clients (not only HomeKit clients if you use http, webserver, websockets together).

jmbwell, 2020-07-05:
Indeed. Lower numbers seem to be more stable but do seem to lead to more "Updating…" and "Not responding" delays in the Home app. Higher numbers seem to improve the experience in the Home app but seem to lead to resets on the ESP. 

For now I guess the best value is for the implementer to determine. Thank you!

closed 2020-07-05 by jmbwell