Mixiaoxiao/Arduino-HomeKit-ESP8266 #36: GarageDoor Example ?

SYNiMAK, 2020-06-27:
Just wondering if there is any chance you could do a garage door opener example? 

P.S. Loving your work!

Mixiaoxiao, 2020-06-27:
This link [garage.c](https://github.com/maximkulkin/esp-homekit-demo/blob/master/examples/garage/garage.c) may help, define your accessory like it and control/report your accessory by reading the Example01~04.

SYNiMAK, 2020-06-28:
Awesome! Thank you. Will give it a go

jmbwell, 2020-06-28:
Here's how I did mine: https://github.com/jmbwell/GargeDoorOpener

askarkurymbayev, 2020-12-12:
you can make garage doors with a wicket function with a door_lock icon?

askarkurymbayev, 2020-12-12:
can you make garage doors with a wicket function with a door_lock icon?

PierreBier, 2021-01-16:
Hello, this is a magnificent project. Thank you!
My Problem is, that i have a RF remote control with only one button to open. After 2 minutes the garage door closes automatically. I don't have access to the sensors.
Is there a way to change the code to auto-close after a delay of 2 minutes?

thncode, 2021-01-17:
> Hello, this is a magnificent project. Thank you!
> My Problem is, that i have a RF remote control with only one button to open. After 2 minutes the garage door closes automatically. I don't have access to the sensors.
> Is there a way to change the code to auto-close after a delay of 2 minutes?

Yes you can use a timer, I do the same with a parking lot barrier...

PierreBier, 2021-01-17:
Thank you for your answer. I guess I have to change the part with the magnetic sensor into a timer. But how do I do that?
Every time I change something with the magnetic sensors the program crashes.