Mixiaoxiao/Arduino-HomeKit-ESP8266 #38: Using an Led Strip with this?

skillsofmatthew, 2020-07-01:
Sorry, I know that this is probably pretty basic, but what wondering how you would go about using an RGB strip with this. I have tried some others but this is the only one that I can actually get to show up on my phone as a device.  

Matt

ETblaky, 2020-07-01:
I am currently working on a project exactly like that. Which part are you asking exactly?

Here's my Accessory.c
```
#include <homekit/characteristics.h>
#include <homekit/homekit.h>

void my_accessory_identify(homekit_value_t _value) {
    //TODO: Blink Led for Identification
}

homekit_characteristic_t cha_on = HOMEKIT_CHARACTERISTIC_(ON, false);
homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "Led Strip");
homekit_characteristic_t cha_bright = HOMEKIT_CHARACTERISTIC_(BRIGHTNESS, 50);
homekit_characteristic_t cha_sat = HOMEKIT_CHARACTERISTIC_(SATURATION, (float) 0);
homekit_characteristic_t cha_hue = HOMEKIT_CHARACTERISTIC_(HUE, (float) 180);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Led Strip"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
        HOMEKIT_SERVICE(LIGHTBULB, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
            &cha_on,
            &cha_name,
	    &cha_bright,
            &cha_sat,
            &cha_hue,
            NULL
        }),
        NULL
    }),
    NULL
};

homekit_server_config_t accessory_config = {
		.accessories = accessories,
		.password = "111-11-111"
};
```

Here's "main homekit file"
```
#include "HomeKitHandler.hpp"
#include "FastLED.h"
#include "../LedHandler.hpp"
#include "presets/Presets.hpp"
#include <Arduino.h>
#include <arduino_homekit_server.h>
#include <debug/Debug.hpp>

#ifndef HOME_KIT_HANDLER_CPP
#define HOME_KIT_HANDLER_CPP

extern "C" homekit_server_config_t accessory_config;

extern "C" homekit_characteristic_t cha_on;
extern "C" homekit_characteristic_t cha_bright;
extern "C" homekit_characteristic_t cha_sat;
extern "C" homekit_characteristic_t cha_hue;

extern Preset current_preset;
extern Solid s;
extern int value;
[... 85 lines cut ...]
}

void IOS::set_bright(const homekit_value_t v) {
    int bright = v.int_value;
    cha_bright.value.int_value = bright; //sync the value
    LED::setValue((bright / 100.0) * 255.0);
}

#endif
```

Just for context, I'm using the FastLed library.
Also, the homekit launches 2 requests when you choose a color: one setting the hue and another one setting the saturation. I preferred to wait for the two requests before updating the strip color, that's what those parts are for:

```
    if((incomplete_color.is_hue_set && !incomplete_color.is_sat_set) || (!incomplete_color.is_hue_set && !incomplete_color.is_sat_set)) {
        incomplete_color = IOS::FullColor();
        incomplete_color.hue = (hue / 360.0) * 255.0;
        incomplete_color.is_hue_set = true;
    } else if(!incomplete_color.is_hue_set && incomplete_color.is_sat_set) {
        if(value == 0) { value = 255; }
        LED::setSolid(CHSV((hue / 360.0) * 255.0, incomplete_color.sat, value), true);
        incomplete_color = IOS::FullColor();
    }
```

skillsofmatthew, 2020-07-02:
To be honest I have no real clue what I’m doing I started working on lighting project because I would be a good thing to start with and learn. I got the simple LED to turn on and off with Siri right now my board isn’t working anymore. I will look at it some more and get back to you. 

Glad to know someone else is working on it though. 

Thanks!

vniehues, 2020-07-31:
@ETblaky Hey There! Could you provide us with your complete example, please? I'm trying to get this to work but there are many things missing and I really can't wrap my head around this stuff.. 
Thanks in advance! :)

ETblaky, 2020-07-31:
Hey @vniehues, here's the complete project: [Wifi Controlled LED](https://github.com/ETblaky/Wifi-Controlled-LED).
The HomeKit part is in `src/handlers/HomeKit/`
Just a small disclaimer: I use different methods to control the strip over wifi (Web Page, MQTT, HomeKit), but not all off them have security measures implemented yet!

vniehues, 2020-07-31:
@ETblaky Thank you! That was really fast.
Unfortunately i'm still having a hard time. But im sure with you code i can get it to work properly.
Thanks again! 💯

ETblaky, 2020-07-31:
No problem! Good luck in your project. Also, if the problem you're facing is regarding the pairing process, I have a tip for you.

vniehues, 2020-08-01:
Thanks! 
Nope, my problems are:
1. understanding the whole c thing...
2. when i managed to run your repo, my led strip was only showing rainbow-colors while using the homekit method. I managed to set it to a solid color directly after start but every change coming from the home app just resulted in it being a weird rainbow...

ETblaky, 2020-08-02:
That's odd, I mean, I did implemented a rainbow preset, but the HomeKit shouldn't be able to set it on.
Maybe try to comment lines 84 and 85 in main.cpp:
```
WEB::loop();
MQTT::loop();
```
That way the only process controlling the led would be the HomeKit.
You could also try debugging `LedHandler.cpp` to check if `current_preset` is set to Solid.

Mixiaoxiao, 2020-10-02:
New `Example05_WS2812_Neopixel` is merged.

closed 2020-10-02 by Mixiaoxiao