Mixiaoxiao/Arduino-HomeKit-ESP8266 #39: Unpredictable pairing with Homekit

pierreauclair, 2020-07-05:
Dear Mixiaoxiao, dear all,

First of all, thanks Mixiaoxiao for your work porting Homekit to ESP8266. This is a fantastic help for DIY hobbyists.

I am using your library (HomeKit-ESP8266@1.2.0) with an ESP8266 on a NodeMCU V3 board and I ‘m having some problems. 

Sorry to bother with what might be a trivial problem. 

For the sake of understanding, I'm using the temperature_sensor example you have provided. 

The process of pairing seems very unpredictable. I could pair the device only once in 20 or 30 attempts.

Most of the time, the process stops at Pair setup step  2 / 3 with a message 
`
>>> [  99510] pair_setup took  26664ms
>>> homekit_client_process: [Client 1073682684] Finished processing
Current temperature: 19.0
Free heap: 39560, HomeKit clients: 1
>>> operator(): MDNS call DynamicServiceTxtCallback
>>> [  99852] HomeKit: [Client 1073682684] Disconnected!
>>> [  99857] HomeKit: [Client 1073682684] Closing client connection
>>> homekit_server_close_client: [Client 1073682684] The sockect is stopped
`
 Sometimes, the log show that the device is paired :

`>>> [  66273] HomeKit: [Client 1073678340] Pair Verify Step 2/2
>>> homekit_server_on_pair_verify: [Client 1073678340] Decrypting payload
>>> homekit_server_on_pair_verify: [Client 1073678340] Searching pairing with 37150F9A-FBD7-447C-8D2D-ED50E86365B8
>>> [  66296] HomeKit: [Client 1073678340] Found pairing with 37150F9A-FBD7-447C-8D2D-ED50E86365B8
>>> homekit_server_on_pair_verify: [Client 1073678340] Verifying device signature
>>> [  66338] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> send_tlv_response: [Client 1073678340] Sending TLV response
>>> client_send: [Client 1073678340] send data size=105, encrypted=false
>>> write: [Client 1073678340] Sending data of size 105
>>> [  67871] HomeKit: [Client 1073678340] Verification successful, secure session established
>>> [  67879] pair_verify took   1606ms
>>> [  67883] HomeKit: Free heap: 41728
>>> homekit_client_process: [Client 1073678340] Finished processing
>>> [  67903] HomeKit: [Client 1073678340] Disconnected!
>>> [  67908] HomeKit: [Client 1073678340] Closing client connection
>>> homekit_server_close_client: [Client 1073678340] The sockect is stopped
>>> operator(): MDNS call DynamicServiceTxtCallback`

But in reality, the HomeKit hub was not able to pair with the accessory. 
My iPhone (iOS 13.5.1) shows "unable to add ... Could not connect to the accessory" (translated from French).

It looks like the connection was somehow broken. Have you ever encountered something like this ? Am I the one one having this problem ?

Thanks in advance for sharing your experience. 
Regards

Mixiaoxiao, 2020-07-05:
It is really strange. Full logs will help maybe (in a txt file, keeping DEBUG log disabled is enough).

pierreauclair, 2020-07-06:
Thanks fro your answer. 
Here are 3 consecutive tries.
- [Homekit_8266_verify_step_1_2.log](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/4876819/Homekit_8266_verify_step_1_2.log) : The process seems to be stopped after pairing in the middle of the verification.

- [Homekit_8266_pair_step_3_3.log](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/4876826/Homekit_8266_pair_step_3_3.log) : The process seems to be working as far as pairing is concerned but then the socket is disconnected.  

- [Homekit_8266_lost_loop.log](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/files/4876835/Homekit_8266_lost_loop.log) : It looks like everything is fine but the iOS device can't still register the accessory.

And sometimes, it works.

Thanks for your help. 

Regards.

Mixiaoxiao, 2020-07-06:
I can't make '...1_2.log' and '...3_3.log' happen on my devices. The '...loop.log' is caused by 'Setup Step 2/3' is too slow that iOS disconnects by timeout, but it happens rarely and you should reboot your ESP to retry.

I guess your problems are caused by your network environment. Maybe you can try 
- another router, e.g. a 2.4G only router (not 5G, not dual band), and make sure your iOS and ESP connect to **same** network. Refer to [issues/12](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/12)
- another ESP board. Some boards may have hardware issues leads to a weak WiFi connection.

pierreauclair, 2020-07-06:
I'll try that. Thanks for your help. 
I'll keep you posted. 
BR

adriankae, 2021-01-08:
> I'll try that. Thanks for your help.
> I'll keep you posted.
> BR

Hie @pierreauclair,

could you solve your issue?

Stealth-ip, 2021-01-20:
i faced this issue in case that esp was connected to laptop with port monitor.
after connection to normal PSU all issues was resolved.