Mixiaoxiao/Arduino-HomeKit-ESP8266 #41: Adding more lights with buttons

makecarafa, 2020-07-11:
Hi! First thank you for your work, it's amazing!

I'm trying to understand how to control up to 3 lights with buttons (like you can do in the simple led example, which has a bug, when you turn off the light from the HomeKit app then you have to press twice the button to turn it on, but I don't care about that issue), the problem is to add more lights with a button for each one, I don't see how the code works and I tried every sketch I found online but everyone is using a different approach so it gets more and more confusing.

On the switch example (which works great) there is no button, I saw you comment "/report the switch value to HomeKit if it is changed (e.g. by a physical button)", but again I tried many things and I couldn't make it work. 

Do you have an example to see how it would work? The final idea of the project is being able to control 3 lights with one button each. I always ended up with a problem identifying each button for each HomeKit service in my accessory.

Thank you so much!

makecarafa, 2020-07-16:
Nobody? nothing? There is no other source for this code online...

ETblaky, 2020-07-18:
I Think the [exemple 4](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/tree/master/examples/Example04_MultipleAccessories) may be able to help you, as each light has to be treated as a new accessory.
As for the problem of having to press the button twice to turn it on, I don't think the bug is from this project, it's more likely something to do with your code.

makecarafa, 2020-07-19:
Thanks for your reply ETblaky!

I already saw the example 4, but I couldn’t make it work with that either. In every example code I see different way he use this code so its kind of confusing how this works.

Regarding the issue with the button, I’m using the example straight as it was published, maybe there is a way to solve it but I couldn’t figured it out either.

Since at least one light is working with one button I will use 3 boards, one for each light and problem solved (kind of).

Now I’m having problems with the board, when its plugged to the pc the code works perfectly, but when I use an external power supply the wifi crashes. I don’t think this is a code problem, must be a power supply issue but already tried every solution posted online I could find and still doesn’t work.

I hope I can make all this work.

Now I’m using the switch example to open the garage door with a custom Siri shortcut, works like a charm... but only when is powered with the pc haha

Thanks guys!

cookier0218, 2020-07-27:
Hi makecarfa,

Would you mind sharing your coding about one light with one button. I'm still struggling in coding. Anyway, thanks a lot.

makecarafa, 2020-07-27:
Hello! The code for one light with one button is the one inside examples/legacy/simple led. I didn’t touch anything and works fine, except for that issue I was talking about in my first message. 

> Hi makecarfa,
> 
> Would you mind sharing your coding about one light with one button. I'm still struggling in coding. Anyway, thanks a lot.