Mixiaoxiao/Arduino-HomeKit-ESP8266 #42: Error homekit_value_t: "sorry, unimplemented  non-trivial designated initializers not supported"

JorgeFrias, 2020-07-12:
When trying to instantiate `HOMEKIT_BOOL(true)` I get the compile error `sorry, unimplemented  non-trivial designated initializers not supported`.

The error is given for all `#define`s to help the initialization of `homekit_value_t` in `types.h`, the lines are: 
``` C++
#define HOMEKIT_BOOL_(value, ...) \
    {.format=homekit_format_bool, .bool_value=(value), ##__VA_ARGS__}
#define HOMEKIT_BOOL(value, ...) (homekit_value_t) HOMEKIT_BOOL_(value, __VA_ARGS__)
```

Seems like the error is given because the initialization of the `homekit_value_t` struct is not given all variables. But maybe not (I do not have profound understanding of C++) I cannot get it to work defining `homekit_value_t` by my self neither with all variables.

If I make any progress I will keep the thread updated.

Thank you

------------------
*Awesome library btw*

JorgeFrias, 2020-07-13:
Seems like this `homekit_value_t ` initialization compiles:
```c++
    homekit_value_t booleanValue = {
        
        .is_null = 0,
        .is_static = 0,
        .format = homekit_format_bool,        
        {
            .bool_value = true   // Nested because is directed to the nested Union in homekit_value_t 
        }
    };
```
Right now I think the initialization macro defined by the library, is not supported by the Arduino compiler.

Just for reference here is the definition of `homekit_value_t`:
``` c++
typedef struct {
    bool is_null : 1;
    bool is_static : 1;
    homekit_format_t format : 6;
    union {
        bool bool_value;
        int int_value;
        uint8_t uint8_value;
        uint16_t uint16_value;
        uint32_t uint32_value;
        uint64_t uint64_value;
        float float_value;
        char *string_value;
        tlv_values_t *tlv_values;
        struct {
            uint8_t *data_value;
            size_t data_size;
        };
    };
} homekit_value_t;
```

Mixiaoxiao, 2020-07-13:
Use `HOMEKIT_BOOL_CPP(true or false)` in .cpp or .ino.

JorgeFrias, 2020-07-13:
Works perfectly. Thank you very much!

closed 2020-07-13 by JorgeFrias