Mixiaoxiao/Arduino-HomeKit-ESP8266 #43: Soft WDT reset

antonioreyna, 2020-07-13:
Hello now im getting this problem and it restarts.

the stack trace says.

```
0x4023c698: _vfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 531
0x4023bf06: vprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vprintf.c line 40
```

```
>>> >> [  65629] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  67145] HomeKit: [Client 1073688468] Verification successful, secure session established
>>> [  67154] HomeKit: Free heap: 28760
>>> [  67158] HomeKit: [Client 1073688436] Pair Verify Step 2/2
>>> [  67165] HomeKit: [Client 1073688436] Found pairing with 
>>> [  67199] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
Soft WDT reset

>>>stack>>>

ctx: cont
```

Mixiaoxiao, 2020-07-13:
Your CPU is 80MHz? Or you has modified the default `user_settings.h`?

antonioreyna, 2020-07-14:
im not sure about the CPU how can I check? its a nodemcu amica esp8266. no modification for user_settings.h

Mixiaoxiao, 2020-07-14:
Your logic code and `arduino_homekit_loop()` in loop cost a long time together which tiggers the WTD. Insert `delay(10)` or `yield` before and after `arduino_homekit_loop()`.

antonioreyna, 2020-07-14:
I added yield() and it keeps rebooting.

```
>>> [  62989] HomeKit: Got new client: local 192.168.0.16:5556, remote 192.168.0.9:51545
>>> [  63018] HomeKit: [Client 1073694052] Pair Verify Step 1/2
>>> [  63636] HomeKit: Free heap: 28112
>>> [  63682] HomeKit: [Client 1073694052] Pair Verify Step 2/2
>>> [  63689] HomeKit: [Client 1073694052] Found pairing with E9E03FDB-3BE6-47F5-A24B-
>>> [  63723] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  65240] HomeKit: [Client 1073694052] Verification successful, secure session established
>>> [  65249] HomeKit: Free heap: 27520
>>> [  65301] HomeKit: [Client 1073694052] Get Accessories
>>> [  65366] HomeKit: Got new client: local 192.168.0.16:5556, remote 192.168.0.19:52335
>>> [  65376] HomeKit: [Client 1073698068] Pair Verify Step 1/2
>>> [  65994] HomeKit: Free heap: 25528
>>> [  66000] HomeKit: [Client 1073694052] Update Characteristics
>>> [  66034] HomeKit: Got new client: local 192.168.0.16:5556, remote 192.168.0.2:49501
>>> [  66044] HomeKit: [Client 1073681492] Pair Verify Step 1/2
>>> [  66662] HomeKit: Free heap: 24136
>>> [  66710] HomeKit: Got new client: local 192.168.0.16:5556, remote 192.168.0.10:49447
>>> [  66720] HomeKit: [Client 1073698628] Pair Verify Step 1/2
>>> [  67338] HomeKit: Free heap: 21384
>>> [  67343] HomeKit: [Client 1073681492] Pair Verify Step 2/2
>>> [  67350] HomeKit: [Client 1073681492] Found pairing with E9E03FDB-3BE6-47F5-A24B-
>>> [  67384] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  68900] HomeKit: [Client 1073681492] Verification successful, secure session established
>>> [  68908] HomeKit: Free heap: 21144
>>> [  68913] HomeKit: [Client 1073698068] Pair Verify Step 2/2
>>> [  68920] HomeKit: [Client 1073698068] Found pairing with E9E03FDB-3BE6-47F5-A24B-
>>> [  68954] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c

Soft WDT reset
```

Mixiaoxiao, 2020-07-14:
Make sure the library is the latest and build your project with CPU@160MHz. Try one of the examples.

antonioreyna, 2020-07-14:
how can i build the projct with CPU@160MHz ?

antonioreyna, 2020-07-14:
got it, apparently is working a 160mhz :D lt me keep testing, is there any issue or something as result of building at 160mhz?

Mixiaoxiao, 2020-07-14:
Faster CPU, less time.

closed 2020-07-18 by Mixiaoxiao