Mixiaoxiao/Arduino-HomeKit-ESP8266 #44: Cannot open header "windows.h" and "winsock2.h" while I'm trying to build on VSCode.

Raiiwhen, 2020-07-14:
I have succeed performer examples on my Nodemcu & iPhone and it totally works. But problem occured while Im trying to build on VSCode. I searched my C:\ fully but can not find these two file.
Why it all worked on ArduinoIDE even without these two files? Am i the only one trying to build on VSCode?

.vscode/c_cpp_properties.json is as follows:
`{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:\\Users\\Administrator\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\**",
                "D:\\Documents\\Arduino\\libraries\\**",
                "C:\\Users\\Administrator\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.2\\**"
            ],
            "forcedInclude": [],
            "intelliSenseMode": "msvc-x64"
        }
    ],
    "version": 4
}`

Mixiaoxiao, 2020-07-14:
Not an issuse about this library. It is about your Arduino config in VS code. Try Google?

Raiiwhen, 2020-07-14:
> Not an issuse about this library. It is about your Arduino config in VS code. Try Google?

I'm trying to install WindowsSDK via VS now to fix it. But still I'm curious why these two headers was included from 
**arduino_homekit_server.h -> crypto.h -> wolfssl/wolfcrypt/ed25519.h -> wolfssl/wolfcrypt/types.h -> wolfssl/wolfcrypt/wc_port.h** line 45~47?
I thought these software has nothing to do with my PC operating system.

Anyway I will keep trying. Thank you so much for your concentration.

Mixiaoxiao, 2020-07-14:
Hi, guy, this library is for Arduino ESP8266 only. The `USE_WINDOWS_API` should NOT be defined here since `_WIN32` does not exist in Arduino platform. WolfSSL can be compiled and used in many platforms, and `_WIN32` is for and exists in MS Windows platform.

Raiiwhen, 2020-07-17:
After servel basic peripherals I found that as a editor, you can just ignore some error such like one in title... all I have to do is just compile(Verify and upload) and it all go fine. After all, thank you so much for your contribution!

closed 2020-07-17 by Raiiwhen