Mixiaoxiao/Arduino-HomeKit-ESP8266 #46: Dimmable Lights

sto-chastic, 2020-07-20:
Hello, again thank you for your amazing work. I was wondering if you have some guidelines for how to control a dimmable light?

Mixiaoxiao, 2020-07-20:
It depends how your lights work. DC or AC, PWM or current control, etc. There is no common way.

sto-chastic, 2020-07-21:
I understand. However, from the configuration `homekit_accessory_t *accessories[]` of the device, for a simple switch, in HomeKit app it only shows two states, how can I configure a dimmer.
If you have an example for a DC dimmer, I would very much appreciate, I think from There I can understand the logic and extend it

Mixiaoxiao, 2020-07-21:
Refer to the HAP doc. Try legcy/simple_led.

sto-chastic, 2020-07-22:
Thank you I will take a look

RobinH6, 2020-12-06:
anything new to make a brightness control with a switch

ApplebaumIan, 2020-12-29:
> Refer to the HAP doc. See legcy/simple_led to know how to define a light_bulb with brightness control.

I'm new to this project, why are all the LED examples legacy?

paullj1, 2021-03-06:
You need to define your switch as a lightbulb instead if you want a dimmer control in HomeKit.

mstaack, 2022-03-12:
this is my code so far, but i only get a switch, no dimming option on ios. any hints?

```
/*
 * my_accessory.c
 * Define the accessory in C language using the Macro in characteristics.h
 *
 *  Created on: 2020-05-15
 *      Author: Mixiaoxiao (Wang Bin)
 */

#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

homekit_characteristic_t cha_on = HOMEKIT_CHARACTERISTIC_(ON, false);
homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "Dimmer");
homekit_characteristic_t cha_bright = HOMEKIT_CHARACTERISTIC_(BRIGHTNESS, 50);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Dimmer"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
		HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
			&cha_on,
			&cha_name,
      &cha_bright,
			NULL
		}),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
};
```

paullj1, 2022-03-12:
Your HOMEKIT_SERVICE needs to be “LIGHTBULB” not SWITCH