Mixiaoxiao/Arduino-HomeKit-ESP8266 #47: TV Example

borissov, 2020-07-21:
Hello, I create simple program to start and stop my TV using IR and simple switch button. I want to make more functions and looking for TV buttons example. Great library! Thanks!

peros550, 2020-07-21:
@borissov why don't you share your code? It sounds very intersting!

borissov, 2020-07-21:
This is test script and is not formated and indented. I get switch example and modified:

```c
#include <Arduino.h>
#include <IRsend.h> //IRremoteESP8266 library
#include <arduino_homekit_server.h> //Arduino-HomeKit-ESP8266 library
#include "wifi_info.h"

#define LOG_D(fmt, ...)   printf_P(PSTR(fmt "\n") , ##__VA_ARGS__);

IRsend irsend(4); 

void setup() {
	Serial.begin(115200);
  
  irsend.begin();
	wifi_connect(); // in wifi_info.h
	//homekit_storage_reset(); // to remove the previous HomeKit pairing storage when you first run this new HomeKit example
	my_homekit_setup();

}

void loop() {
	my_homekit_loop();
	delay(10);
}

//==============================
// HomeKit setup and loop
//==============================

// access your HomeKit characteristics defined in my_accessory.c
extern "C" homekit_server_config_t config;
extern "C" homekit_characteristic_t cha_switch_on;

static uint32_t next_heap_millis = 0;

//#define PIN_SWITCH 2

//Called when the switch value is changed by iOS Home APP
void cha_switch_on_setter(const homekit_value_t value) {
	bool on = value.bool_value;
	cha_switch_on.value.bool_value = on;	//sync the value
	LOG_D("Switch: %s", on ? "ON" : "OFF");

  if(on)
  {
    irsend.sendNEC(551494620, 32); //on
  }
  else
  {
    irsend.sendNEC(551527260, 32); //off
  }
	//digitalWrite(PIN_SWITCH, on ? LOW : HIGH);
}

void my_homekit_setup() {
//   irsend.sendNEC(request.toInt(), 32);
//	pinMode(PIN_SWITCH, OUTPUT);
	//digitalWrite(PIN_SWITCH, HIGH);

	//Add the .setter function to get the switch-event sent from iOS Home APP.
	//The .setter should be added before arduino_homekit_setup.
	//HomeKit sever uses the .setter_ex internally, see homekit_accessories_init function.
	//Maybe this is a legacy design issue in the original esp-homekit library,
	//and I have no reason to modify this "feature".
	cha_switch_on.setter = cha_switch_on_setter;
	arduino_homekit_setup(&config);

	//report the switch value to HomeKit if it is changed (e.g. by a physical button)
	//bool switch_is_on = true/false;
	//cha_switch_on.value.bool_value = switch_is_on;
	//homekit_characteristic_notify(&cha_switch_on, cha_switch_on.value);
}

void my_homekit_loop() {
	arduino_homekit_loop();
	const uint32_t t = millis();
	if (t > next_heap_millis) {
		// show heap info every 5 seconds
		next_heap_millis = t + 5 * 1000;
		LOG_D("Free heap: %d, HomeKit clients: %d",
				ESP.getFreeHeap(), arduino_homekit_connected_clients_count());

	}
}

```
Codes for LG TV: https://gitlab.com/snippets/1690600

Mixiaoxiao, 2020-07-22:
There is no TV Services/Characteristics defined in Apple's HAP. So what you are looking for is impossible currently.

Mixiaoxiao, 2020-07-24:
Not an issue about this library, will be closed soon.

closed 2020-07-25 by Mixiaoxiao