Mixiaoxiao/Arduino-HomeKit-ESP8266 #50: Unable to pair with garage characteristic

KochC, 2020-07-29:
Hi, 

great library first of all. I tried to make an garage opener with `current_state`, `target_state`, `obstruction_detected`, `current_lock` and `target_lock` as described in the library. But I can not connect to the device.

The messages I get from the controller seem to look successful, but on the phone I get the `Couldn't add Device` `Home couldn't connect to this accessory.`.
```
>>> [  37600] HomeKit: [Client 1073682956] Pair Setup Step 1/3
Free heap: 40648, HomeKit clients: 1
Report!
Free heap: 40584, HomeKit clients: 1
Free heap: 40536, HomeKit clients: 1
>>> [  51412] HomeKit: [Client 1073682956] Pair Setup Step 2/3
>>> [  51428] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [  68754] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
Free heap: 40184, HomeKit clients: 1
Report!
>>> [  77643] HomeKit: [Client 1073682956] Pair Setup Step 3/3
>>> [  77693] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  79191] HomeKit: Added pairing with 9530CB6F-DC47-47BE-8DA3-A19A2CEEC876
>>> [  79310] HomeKit: Free saved_preinit_pairing_context
>>> [  79315] HomeKit: [Client 1073682956] Successfully paired
>>> [  79382] HomeKit: [Client 1073682956] Disconnected!
>>> [  79387] HomeKit: [Client 1073682956] Closing client connection
>>> [  79434] HomeKit: Got new client: local 192.168.178.24:5556, remote 192.168.178.31:62810
>>> [  79444] HomeKit: [Client 1073678596] Pair Verify Step 1/2
>>> [  80062] HomeKit: Free heap: 42304
>>> [  80137] HomeKit: [Client 1073678596] Pair Verify Step 2/2
>>> [  80144] HomeKit: [Client 1073678596] Found pairing with 9530CB6F-DC47-47BE-8DA3-A19A2CEEC876
>>> [  80179] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [  81695] HomeKit: [Client 1073678596] Verification successful, secure session established
>>> [  81704] HomeKit: Free heap: 42416
>>> [  81820] HomeKit: [Client 1073678596] Get Accessories
>>> [  81947] HomeKit: [Client 1073678596] Disconnected!
>>> [  81953] HomeKit: [Client 1073678596] Closing client connection
```

Here is my accessory file, maybe there is some issues with that:
```
#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value)
{
    printf("accessory identify\n");
}

homekit_characteristic_t cha_garage_name = HOMEKIT_CHARACTERISTIC_(NAME, "Garage");
homekit_characteristic_t cha_garage_current_state = HOMEKIT_CHARACTERISTIC_(CURRENT_DOOR_STATE, 0);
homekit_characteristic_t cha_garage_target_state = HOMEKIT_CHARACTERISTIC_(TARGET_DOOR_STATE, 0);
homekit_characteristic_t cha_garage_obstruction = HOMEKIT_CHARACTERISTIC_(OBSTRUCTION_DETECTED, false);
homekit_characteristic_t cha_lock_current_state = HOMEKIT_CHARACTERISTIC_(LOCK_CURRENT_STATE, 0);
homekit_characteristic_t cha_lock_target_state = HOMEKIT_CHARACTERISTIC_(LOCK_TARGET_STATE, 0);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(
            .id = 2, .category = homekit_accessory_category_garage, .services = (homekit_service_t *[]){
                HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics = (homekit_characteristic_t *[]){
                    HOMEKIT_CHARACTERISTIC(NAME, "Garage"), 
                    HOMEKIT_CHARACTERISTIC(MANUFACTURER, "com.de"), 
                    HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0000002"), 
                    HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266"), 
                    HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify), 
                    NULL}), 
                HOMEKIT_SERVICE(GARAGE_DOOR_OPENER, .primary = true, .characteristics = (homekit_characteristic_t *[]){
                    &cha_garage_current_state, 
                    &cha_garage_target_state, 
                    &cha_garage_obstruction, 
                    &cha_garage_name, 
                    &cha_lock_current_state, 
                    &cha_lock_target_state, 
                    NULL}), 
                NULL}),
    NULL};

homekit_server_config_t config = {
    .accessories = accessories,
    .password = "111-11-111"};

```

Any ideas? Please let me know how and if you need more detailed log files.

Mixiaoxiao, 2020-07-29:
Generally caused by invalid cha-value(e.g. out of range). Check the HAP defination and your code.

jmbwell, 2020-07-29:
I would add that it looks like pairing is stopping after step 2/3. When mine do that I keep retrying and it will pair eventually. Make sure you're at 160Mhz also.

KochC, 2020-07-30:
Thanks to @Mixiaoxiao: I sent `bool` values to `uint8` characteristics, obviously wrong. Changed it to `uint8`, works like a charm. Thanks a lot.

closed 2020-07-30 by KochC