Mixiaoxiao/Arduino-HomeKit-ESP8266 #51: Problem wirt connecting to the esp

Phenixdv, 2020-07-29:
HI,
I’m Having trouble with connecting to my ESP 32, I uploaded the switch example sketch provided,
but on the phone, it doesn’t show up to select to pair with it. (I also tried it with the Temperature sensor and it didn’t work either)
these are the messages I get from the ESP,

[image]

it functioned a few times before until I uncomment the command; `homekit_storage_reset()`
in the message `>>> [ 343] HomeKit: Found admin pairing with A1D6497F-4683-49CF-935B-2321A9ABB508, disabling pair setup`
it says `disabling pair setup` I don’t know if that does mean anything but it doesn’t sound right to me that early.

Mixiaoxiao, 2020-07-30:
The paired info is saved in Flash. You should call `homekit_storage_reset()` or erase the full flash before you upload a new project. The paired esp (which has paired info in Flash) will disable the new pair.
See [https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/10#issuecomment-613969645](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/10#issuecomment-613969645)

Phenixdv, 2020-07-30:
I did, I also flashed an empty scatch on the ESP but nothing changed.

Mixiaoxiao, 2020-07-30:
Your `flashed an empty scatch(->sketch)` does **NOT** equal to `erase the full flash`. IDE settings `Erase Flash`: select All Flash Contents. Or use ESP flash tool to erase full flash.

Phenixdv, 2020-07-30:
thanks for the help, I used the esp 32 download tool, to erase the full flash and it workes now!

closed 2020-07-30 by Phenixdv