Mixiaoxiao/Arduino-HomeKit-ESP8266 #56: Can I run other code in the loop?

DavideVito, 2020-09-12:
Hi, I’m impressed by the simplicity of this project and I was wondering, is it possible to add other code in the main loop function and execute it?

This question raises because I’ve noticed that you call `arduino_homekit_loop()` function so I think that an instruction placed on the sketch loop will be ignored or run only one time.

waaraawa, 2020-09-12:
I'm calling arduino_homekit_loop() once every 500 msec. I believe you can add your code in the loop() additionally. Just you make sure to call arduino_homekit_loop() periodically.

Mixiaoxiao, 2020-10-02:
Yes, you can. But you should remember that you must call `arduino_homekit_loop()` frequently with as small delay as possible. Otherwise you will have a delayed response or unreachable issue in your Home app.

closed 2020-10-20 by Mixiaoxiao