Mixiaoxiao/Arduino-HomeKit-ESP8266 #57: AC control

CommyD, 2020-09-14:
hi all, awesome work on this. 👍 
is there a way to change the accessory in order to work as an ON/OFF button to an AC?

cheers

waaraawa, 2020-09-16:
Did you try it with relay?
https://en.wikipedia.org/wiki/Relay

CommyD, 2020-09-16:
Thanks for the input, but my thought was to use an IR transmitter added to my Wemos D1 mini, and change this code to be seen as a switch in Home.app rather than a light. By having a light accessory, it will also run my AC when I would say “siri, turn off the lights” :) 

Best

waaraawa, 2020-09-16:
Yes you can control any homekit DIY devices by Siri. And you can add your Wemos like AC.
You can choose one or more Homekit characteristics of your Wemos as follow link.
https://developer.apple.com/documentation/homekit/hmcharacteristic/characteristic_types
HOMEKIT_SERVICE_HEATER_COOLER or HOMEKIT_SERVICE_THERMOSTAT are could be used for AC.

I also making AC controller + AQI + temp _humi with this repo. Not finished yet but most functions are working well now:)
[image]
Happy DIY~

jmbwell, 2020-09-16:
If you want your Wemos to be the controller rather than the thing-being-controlled, perhaps you want to implement a “stateless programmable switch”

jmbwell, 2020-09-16:
At first I think I understood the opposite of what you want to do. If you just want on and off, then perhaps all you need to implement is the Outlet service.

CommyD, 2020-09-16:
hi @waaraawa @jmbwell, yes, indeed, just wanted an on/off switch, as i'm not too advanced to work on more complicated things yet. :)
i've taken the Example02_Switch and added the IR hex code in cha_switch_on_setter() method and it works like a charm. 

best regards

CommyD, 2020-09-19:
hi guys,

i have a small logic error in my Switch program, and I would appreciate if someone can help.

//Called when the switch value is changed by iOS Home APP
void cha_switch_on_setter(const homekit_value_t value) {
	bool on = value.bool_value;
	cha_switch_on.value.bool_value = on;	//sync the value
  LOG_D("Switch: %s", on ? "ON" : "OFF");
  digitalWrite(IR_SEND_PIN, on ? LOW : HIGH);
  irsend.sendRaw(rawData, 100, 38); 
}

As seen above, in cha_switch_on_setter method, i've added the IR code to be sent to my AC; this works fine to turn on/off the AC, but unfortunately, if I put this in an automation, it sends the the code regardless of the state of the switch.

Example: I have an automation to turn off my lights and my AC if I leave my house. In this case, when i leave home, if the AC is already off, the state of the switch remains the same, but it also sends an IR signal to my AC (so actually the AC turns on). :|

thanks

CommyD, 2020-09-24:
Hello, wondering if someone can help me out? :)

Thanks

peros550, 2020-09-24:
for ideas on the logic, you may want to have a look in this [example ](https://github.com/peros550/esp-homekit-multiple-sensors/tree/master/examples/multiple_sensors_ac)which works pretty much ok.

CommyD, 2020-09-28:
thank you @peros550  :)