Mixiaoxiao/Arduino-HomeKit-ESP8266 #58: Pair error

CommyD, 2020-09-15:
Hi guys,

I was using this code with Wemos D1 Mini and it worked, but after trying different changes, it stopped pairing and i can't see the accessory in the Home.app; even the raw code doesn't work now. :|

I get this in the Serial Monitor:
>>> [    115] HomeKit: Found admin pairing with 7D052499-30E6-4193-9FBD-C62FBF662DC7, disabling pair setup
>>> [    124] HomeKit: Configuring MDNS
>>> [    127] HomeKit: Init server over
>>> [   1331] HomeKit: heap: 47472, sockets: 0

Can someone help please?

thanks

waaraawa, 2020-09-16:
I guess your auth info is stored inside nvram of your wemos.
Please check "Arduino IDE -> Tools -> Erase Flash" is "Sketch only" or not.
And upload again after change it to "All flash contents".

CommyD, 2020-09-16:
it worked, thank you :)

closed 2020-09-16 by CommyD