Mixiaoxiao/Arduino-HomeKit-ESP8266 #59: how to start with this?

thncode, 2020-09-24:
Sorry for a possibly stupid question: I did already some successful Arduino based projects, but I am not able to make this one compile / run for the ESP8266, how do I compile the sources with the Arduino IDE? Do I have to put that in a specific order form into the IDE library?

THX !

YaoGi, 2020-09-24:
我在 bilibili上看到有帮助的视频你可以看看 https://www.bilibili.com/video/BV1sK411n7Qd?from=search&seid=15486674027743520394

YaoGi, 2020-09-24:
I saw a helpful video on bilibilibili, you can see it https://www.bilibili.com/video/BV1sK411n7Qd?from=search&seid=15486674027743520394

thncode, 2020-09-24:
thx so far

I do get following error messages (didn't do any changes to the sources):

Arduino: 1.8.12 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o:(.text.wc_SrpSetKeyH+0x0): undefined reference to `wc_InitSha512'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o:(.text.wc_SrpSetKeyH+0x4): undefined reference to `wc_Sha512Update'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o:(.text.wc_SrpSetKeyH+0x8): undefined reference to `wc_Sha512Final'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o: in function `wc_SrpSetKeyH':
/Users/thomas/Documents/Arduino/libraries/src/crypto.c:64: undefined reference to `wc_InitSha512'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /Users/thomas/Documents/Arduino/libraries/src/crypto.c:70: undefined reference to `wc_Sha512Update'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /Users/thomas/Documents/Arduino/libraries/src/crypto.c:71: undefined reference to `wc_Sha512Final'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /Users/thomas/Documents/Arduino/libraries/src/crypto.c:77: undefined reference to `wc_SrpInit'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o: in function `crypto_srp_new':
/Users/thomas/Documents/Arduino/libraries/src/crypto.c:84: undefined reference to `wc_SrpInit'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /Users/thomas/Documents/Arduino/libraries/src/crypto.c:92: undefined reference to `wc_SrpTerm'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o: in function `crypto_srp_free':
/Users/thomas/Documents/Arduino/libraries/src/crypto.c:95: undefined reference to `wc_SrpTerm'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o:(.text.crypto_srp_init+0x8): undefined reference to `wc_SrpSetUsername'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o:(.text.crypto_srp_init+0xc): undefined reference to `wc_SrpSetParams'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o:(.text.crypto_srp_init+0x10): undefined reference to `wc_SrpSetPassword'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o:(.text.crypto_srp_init+0x14): undefined reference to `wc_SrpGetVerifier'
Mehrere Bibliotheken wurden für "WiFiServer.h" gefunden
 Benutzt: /Users/thomas/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o:(.text.crypto_srp_init+0x18): undefined reference to `wc_SrpSetVerifier'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/crypto.c.o: in function `crypto_srp_init':
/Users/thomas/Documents/Arduino/libraries/src/crypto.c:101: undefined reference to `wc_SrpSetUsername'
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /Users/thomas/Documents/Arduino/libraries/src/crypto.c:109: undefined reference to `wc_SrpSetParams'
[... cut ...]
/Users/thomas/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/src/arduino_homekit_server.cpp.o: in function `homekit_mdns_init(homekit_server_t*)':
/Users/thomas/Documents/Arduino/libraries/src/arduino_homekit_server.cpp:3176: undefined reference to `wc_Sha512Hash'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

closed 2020-09-24 by thncode