Mixiaoxiao/Arduino-HomeKit-ESP8266 #63: Cannot add same accessory after removing

MalinduDilshan, 2020-10-02:
Hi All,

I have tried the simple_led example and its worked as expected. But suddenly i try to remove it from HomeKit and I tried to add same accessory again, but it won't show in the HomeKit as very first time. I try to change config setupId and password but it also failed. Do you have any suggestions?

Mixiaoxiao, 2020-10-02:
See [https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/10#issuecomment-613969645](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/issues/10#issuecomment-613969645)

Mixiaoxiao, 2020-10-02:
You should:
1. read and use the Example01_TemperatureSensor with detailed comments
    to know the basic concept and usage of this library before other examples。
2. erase the full flash or call homekit_storage_reset() in setup()
    to remove the previous HomeKit pairing storage and
    enable the pairing with the new accessory of new HomeKit project.

MalinduDilshan, 2020-10-02:
Yeah Its working. Thanks. Btw can't we use apple WAC (wireless accessory configuration) instead hard coding ssid and password for the wifi?

closed 2020-10-02 by MalinduDilshan