Mixiaoxiao/Arduino-HomeKit-ESP8266 #64: Custom unit values?

thncode, 2020-10-05:
Did anybody find out how to specify custom units for custom characteristics?

thncode, 2020-10-06:
ok - found it by myself...

closed 2020-10-06 by thncode

vtpr, 2020-11-27:
Can you tell me more about it?

thncode, 2020-11-28:
you need to extend the switch statement in Arduino_homekit_server.cpp:

		const char *unit_str = NULL;
		switch (ch->unit) {
		case homekit_unit_none:
			break;
		case homekit_unit_celsius:
			unit_str = "celsius";
			break;
		case homekit_unit_percentage:
			unit_str = "percentage";
			break;
		case homekit_unit_arcdegrees:
			unit_str = "arcdegrees";
			break;
		case homekit_unit_lux:
			unit_str = "lux";
			break;
		case homekit_unit_seconds:
			unit_str = "seconds";
			break;
		case homekit_unit_kmh:
			unit_str = "km/h";
			break;
		case homekit_unit_mm:
			unit_str = "mm";
			break;
		case homekit_unit_wm2:
			unit_str = "W/m2";
			break;
		}