Mixiaoxiao/Arduino-HomeKit-ESP8266 #66: Help getting relay to work

brennants2658, 2020-10-06:
I've spent hours upon hours trying to find out how to get the toggle in the home app to open/close my relay. I must be missing something obvious. What code should I be adding where to get this to work?

shibbidydibbidy, 2020-10-20:
Just use simplest_led it is basically plug and play. Just set your PIN_LED to the pin you are using to control your relay.

alperenunuvar, 2020-12-07:
@shibbidydibbidy is it possible more than one relay to add? how?

shibbidydibbidy, 2020-12-08:
> @shibbidydibbidy is it possible to control multiple relay? how?

Yeah just connect both relays in parallel. Use the same PIN_LED for both.

alperenunuvar, 2020-12-08:
Thank you very much for your reply, but if we use the same PIN_LED for both parallel relays, when I try to close one of them, are both of them closed? Could you explain it?

alperenunuvar, 2020-12-08:
> > @shibbidydibbidy is it possible to control multiple relay? how?
> 
> Yeah just connect both relays in parallel. Use the same PIN_LED for both.

Thank you very much for your reply, but if we use the same PIN_LED for both parallel relays, when I try to close one of them, are both of them closed? Could you explain it?