Mixiaoxiao/Arduino-HomeKit-ESP8266 #67: Thermostat giving error “Accessory out of compliance” when trying to pair

daz, 2020-10-10:
All other accessories and examples I try work fine, but thermostat is giving me the error. 

```c
#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
  printf("accessory identify\n");
}

homekit_characteristic_t cha_current_heating_cooling_state = HOMEKIT_CHARACTERISTIC_(CURRENT_HEATING_COOLING_STATE, 0);
homekit_characteristic_t cha_target_heating_cooling_state = HOMEKIT_CHARACTERISTIC_(TARGET_HEATING_COOLING_STATE, 0);
homekit_characteristic_t cha_current_temperature = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0);
homekit_characteristic_t cha_target_temperature = HOMEKIT_CHARACTERISTIC_(TARGET_TEMPERATURE, 0);
homekit_characteristic_t cha_temperature_display_units = HOMEKIT_CHARACTERISTIC_(TEMPERATURE_DISPLAY_UNITS, 0);

homekit_accessory_t *accessories[] = {
  HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_thermostat, .services=(homekit_service_t*[]) {
    HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Thermostat"),
      HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
      HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
      HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
      HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
      HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
      NULL
    }),
    HOMEKIT_SERVICE(THERMOSTAT, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
      &cha_current_heating_cooling_state,
      &cha_target_heating_cooling_state,
      &cha_current_temperature,
      &cha_target_temperature,
      &cha_temperature_display_units,
      NULL
    }),
    NULL
  }),
  NULL
};

homekit_server_config_t config = {
  .accessories = accessories,
  .password = "111-11-111"
};
```

Any ideas?

waaraawa, 2020-10-10:
Hi daz,
You should make homekit can set cooling status, temperature and temperature unit.
here is mine thermostat definitions.
`
// Thermostat Service, HOMEKIT_SERVICE_THERMOSTAT
homekit_characteristic_t h_thermo_c_st = HOMEKIT_CHARACTERISTIC_(CURRENT_HEATING_COOLING_STATE, 0);
homekit_characteristic_t h_thermo_t_st = HOMEKIT_CHARACTERISTIC_(TARGET_HEATING_COOLING_STATE, 0, .getter = get_target_heating_cooling_state, .setter = set_target_heating_cooling_state);
homekit_characteristic_t h_thermo_c_temp = HOMEKIT_CHARACTERISTIC_(CURRENT_TEMPERATURE, 0);
homekit_characteristic_t h_thermo_t_temp = HOMEKIT_CHARACTERISTIC_(TARGET_TEMPERATURE, 0, .getter = get_target_temperature, .setter = set_target_temperature);
homekit_characteristic_t h_thermo_temp_unit = HOMEKIT_CHARACTERISTIC_(TEMPERATURE_DISPLAY_UNITS, 0, .getter = get_temperature_display_unit, .setter = set_temperature_display_unit);
homekit_characteristic_t h_thermo_c_humi = HOMEKIT_CHARACTERISTIC_(CURRENT_RELATIVE_HUMIDITY, 0);
`
also you can check whole codes as following my code.
https://github.com/waaraawa/Homekit_AQI_Desk/blob/master/homekit_config.c

BR

daz, 2020-10-11:
When adding the setters/getters I'm still getting an error "Accessory is out of compliance".

Same error when I copy your code verbatim. But when I remove this section, it pairs:

```c
            HOMEKIT_SERVICE(
                THERMOSTAT,
                .primary = true,
                .characteristics = (homekit_characteristic_t*[]) {
                    HOMEKIT_CHARACTERISTIC(NAME, "Air Conditioner"),
                    &h_thermo_c_st,
                    &h_thermo_t_st,
                    &h_thermo_c_temp,
                    &h_thermo_t_temp,
                    &h_thermo_temp_unit,
                    &h_thermo_c_humi,
                    NULL
                }
            ),
```

waaraawa, 2020-10-11:
@daz 
If you can share your homekit_config.c file, I can review it but I don't know why with not enought info.
I hope you review which info you should make get/set as characteristics.h

`/**
 Defines that the accessory contains a thermostat.

 Required Characteristics:
 - CURRENT_HEATING_COOLING_STATE
 - TARGET_HEATING_COOLING_STATE
 - CURRENT_TEMPERATURE
 - TARGET_TEMPERATURE
 - TEMPERATURE_DISPLAY_UNITS
 
 Optional Characteristics:
 - NAME
 - CURRENT_RELATIVE_HUMIDITY
 - TARGET_RELATIVE_HUMIDITY
 - COOLING_THRESHOLD_TEMPERATURE
 - HEATING_THRESHOLD_TEMPERATURE
 */
#define HOMEKIT_SERVICE_THERMOSTAT HOMEKIT_APPLE_UUID2("4A")`

daz, 2020-10-15:
Okay I found the problem, the default target temperature variable was zero, which is out of range.

closed 2020-10-15 by daz

andynovak12, 2022-05-28:
> Okay I found the problem, the default target temperature variable was zero, which is out of range.

I had a similar problem with the CURRENT_TEMPERATURE. However for me, the issue wasn't with the default value (I had it correctly set to 20). 

My CURRENT_TEMPERATURE value was being overwritten by a temperature sensor's values, and at the time of pairing with Homekit, the sensor was unplugged and setting CURRENT_TEMPERATURE to a value outside of the "acceptable" temperature range defined by Apple.

Hopefully this helps someone in the future if they are ever in the situation I was in.

korgoth, 2022-11-05:
Hey guys, 

im struggling with creating a thermostat also - either it doesnt pair or im getting stuck on "updating" state. 

Can someone provide a working example of how they setup their thermostat using this library? 

Any help is much appreciated!

mateusmsantin, 2023-02-23:
look it in your code:
HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_thermostat, .services=(homekit_service_t[]) {
HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_thermostat, .services=(homekit_service_t*****[]) {

ericying, 2024-09-30:
Very late, but hope this helps... I ran into the exact same problem and turned out I was using the wrong mandatory characteristics that are similarly named but with different UUIDs from characteristics.h

For air-conditioner/heater-cooler, the ones to use are:
HOMEKIT_CHARACTERISTIC_CURRENT_HEATER_COOLER_STATE
HOMEKIT_CHARACTERISTIC_TARGET_HEATER_COOLER_STATE

NOT to be confused with:
HOMEKIT_CHARACTERISTIC_CURRENT_HEATING_COOLING_STATE
HOMEKIT_CHARACTERISTIC_TARGET_HEATING_COOLING_STATE
I haven't looked into whether these are deprecated or meant for another accessory but they're the cause of Updating... in the Home app.

So, just edit your my_accessories.c to
// 0: Off, 1: Heat, 2: Cool
homekit_characteristic_t ac_current_state = HOMEKIT_CHARACTERISTIC_(CURRENT_HEATER_COOLER_STATE, 0);

// 0: off, 1: heat, 2:cool, 3: auto
homekit_characteristic_t ac_target_state = HOMEKIT_CHARACTERISTIC_(CURRENT_HEATER_COOLER_STATE, 0);