Mixiaoxiao/Arduino-HomeKit-ESP8266 #70: Not active?

thncode, 2020-10-14:
That's strange: although I do have an active characteristic with default true the service is reported as "inactive", anyone any idea?

Cheers, 
 Thomas

// (optional) format: bool; HAP section 9.96
homekit_characteristic_t cha_status_active = HOMEKIT_CHARACTERISTIC_(STATUS_ACTIVE, true);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_bridge, .services=(homekit_service_t*[]) {
      HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "WeatherStation"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "n/a"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "000001"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
        NULL
    }),
  HOMEKIT_ACCESSORY(.id=2, .category=homekit_accessory_category_sensor, .services=(homekit_service_t*[]) {
      HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "WeatherStation"),
      HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
      NULL
    }),
      HOMEKIT_SERVICE(TEMPERATURE_SENSOR, .primary=true, .characteristics=(homekit_characteristic_t*[]) {
      HOMEKIT_CHARACTERISTIC(NAME, "Temperature"),
      &cha_temperature,
      &cha_status_active,
      &cha_humidity,
      &cha_light,
      &cha_uv,
      &cha_air_pressure,
      &cha_avg_wind,
      &cha_max_wind,
      &cha_rain1h,
      &cha_rain_bool,
      &cha_rain_day,
      &cha_storm,
      &cha_air_quality,
      &cha_air_particulate_density,
      &cha_air_particulate_size,
      &cha_battery_level,
      NULL
    }),
    NULL
  }),
  NULL
};

thncode, 2020-10-25:
seems to be a problem with accessories of bridges...

sircuri, 2020-12-14:
You need to implement the getter on the homekit_characteristic_t as well to allow your ios device to get the latest status.