Mixiaoxiao/Arduino-HomeKit-ESP8266 #71: MDNS.announce

CommyD, 2020-10-15:
Hi everyone,

I've been using this library for a while, and i often see that the accessory is not responding; looking thru the closed issues, i see that by using MDNS.announce(), could be a possible fix. 
Can someone please let me know how to use/ where to write MDNS.announce()? a printscreen would really help :)

thanks in advance

emelianov, 2020-10-17:
Hello @CommyD 
Try to add this code:

```c
#include <Schedule.h>

bool announce() {
  MDNS.announce();
  return true;
}

void setup() {
//...
  schedule_recurrent_function_us(announce, 5000000);
//...
}
```

ayush9upta, 2022-06-17:
@emelianov hello sir, i dont know if you'll respond to this, but i don't know where to add this following code into, is it into the main code or somewhere else, i've ESP32 and i know you said this code isn't applicable to that, But please help me it means a-lot!