Mixiaoxiao/Arduino-HomeKit-ESP8266 #73: Can't find accessory in Home app

slash-se, 2020-10-18:
Hello!
I'am try to use your library with example sketch **simple_led.ino** in Arduino IDE.
Only change I made was _ssid_ and _password_
Also Iam forrlw your recomendation in Arduino IDE settings like CPU Frequency, SSL Support etc.
After flash ESP8266 in Serial Monitor I get messages

`SketchSize: 480432
FreeSketchSpace: 1613824
FlashChipSize: 4194304
FlashChipRealSize: 4194304
FlashChipSpeed: 40000000
SdkVersion: 2.2.2-dev(38a443e)
FullVersion: SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
CpuFreq: 160MHz
FreeHeap: 46696
ResetInfo: External System
ResetReason: External System
>>> [     99] HomeKit: Free heap: 46696
OFF
>>> [    104] HomeKit: Starting server
>>> [    107] HomeKit: Using existing accessory ID: A2:B2:3C:56:CE:B9
>>> [    114] HomeKit: Preiniting pairing context
>>> [    118] HomeKit: Using user-specified password: 111-11-111
>>> [    140] HomeKit: Call s_mp_exptmod in integer.c, original winsize 6
>>> [   6197] HomeKit: Call s_mp_exptmod in integer.c, original winsize 5
>>> [   9279] HomeKit: Preinit pairing context success
>>> [   9284] HomeKit: Configuring MDNS
>>> [   9287] HomeKit: Init server over
>>> [   9291] HomeKit: Free heap: 43736
heap: 43736, sockets: 0
>>> [  12884] HomeKit: WiFi connected, ip: 192.168.1.241, mask: 255.255.255.0, gw: 192.168.1.1
>>> [  12893] HomeKit: Configuring MDNS
>>> [  12898] HomeKit: MDNS begin: ESP8266_LED_5FB665, IP: 192.168.1.241
heap: 41904, sockets: 0
heap: 41904, sockets: 0
heap: 41400, sockets: 0
heap: 41400, sockets: 0
heap: 41400, sockets: 0
heap: 41400, sockets: 0
heap: 41400, sockets: 0
heap: 41400, sockets: 0`

But I can't find accessory in my Home app on iPhone or iPad.
Can you please tell me what I'am doing wrong or what I need to change to connect to ESP8266 in Home app?
Also I try **simplest_led.ino** from examples but don't reach success.

Mixiaoxiao, 2020-10-18:
Your iPhone and ESP are in same WiFi? (both in 2.4G?)

slash-se, 2020-10-18:
@Mixiaoxiao Yes both are in the same wi-fi network. 2.4GHz

Mixiaoxiao, 2020-10-18:
Something wrong with your router's settings. MDNS broadcast does NOT work on your LAN network.

slash-se, 2020-10-18:
Maybe you know some way to check if it's problem with my router and mDNS?

slash-se, 2020-10-18:
@Mixiaoxiao Did mDNS need only with Arduino IDE? 
I try to flash my ESP8266 with [esp-homekit](https://github.com/maximkulkin/esp-homekit) and everything going good at my Wi-Fi network.

sircuri, 2020-12-13:
Maybe an old topic, but did you pair the device in the past already? Looking at the log provided it is reusing an existing accessory id. Try reset the homekit storage by calling homekit_storage_reset() in your setup or erase the flash before flashing it again. Might help.