Mixiaoxiao/Arduino-HomeKit-ESP8266 #77: Accessory not responding

thncode, 2020-10-25:
I do now have five accessories in place (nice library btw!!!) but one, a weather station does always react unresponsive after some hours. I optimized threading and sequencing as much as I could, but what lets HomeKit assume an accessory does not respond? my accessories all do have web interfaces, and they are very responsive via http...

thncode, 2020-10-28:
ok, now I found out that this happens only if there are custom characteristics defined. If I stay with stock characteristics, the accessory works responsive....

thncode, 2020-10-28:
ok: that's funny: whenever I use .description statements in characteristics definitions like

homekit_characteristic_t cha_max_wind = HOMEKIT_CHARACTERISTIC_(CUSTOM,
    .description = "Böen",
    .type = "1b3d4324-9d68-11e8-9d55-f7a461994af7",
    .format = homekit_format_float,
    .unit = homekit_unit_kmh,
    .min_value = (float[]) {0.0},
    .max_value = (float[]) {300.0},
    .min_step = (float[]) {0.1}, 
    .permissions = homekit_permissions_paired_read | homekit_permissions_notify
);

 the accessory does not want to connect resp. it's pure luck if a connection is possible. Whenever I leave that out it behaves correctly. But I need those descriptions for custom characteristics! @Mixiaoxiao any idea?

hdirand, 2020-11-21:
I have a similar issue, when using siri (with my homepod), accesory doesn’t respond, but when I use homkit app on my smartphone, accessory is doing well...

hdirand, 2020-11-23:
My issue is now fixed and All is ok. I’ve read another post in another issue and applied the proposed solution : I’ve add MDNS.announce() every 5s, and it is working well. Thanks for your job @Mixiaoxiao

d4rkmen, 2020-11-23:
Please consider, this workaround makes mass of undesired UDP traffic in the local network. More accessories - more garbage traffic.

hdirand, 2020-11-23:
Ok thks @d4rkmen , that’s the reason why I add MDNS.announce each 5s. Do you have a best solution ?

Sanchikuuus, 2021-05-28:
Have same issue. 
Try like said @hdirand with MSDN.announce() but it doesn’t help. 
Any ideas?

jenspr, 2021-07-11:
Any updates here?
I am suffering from same... It says disconnected in Homekit. (Though WIFI fine, I can ping the device).
I have no clue what is causing that and how to prevent. Any help appreciated. 
Its so said as this is a brilliant library/Project!!!! ... will it work better with an ESP32?