Mixiaoxiao/Arduino-HomeKit-ESP8266 #79: Need help with Example05 (WS2812) and existing webserver

SchneHa, 2020-10-30:
I tried to combine an existing project (controlling WS2812 via webserver) and Example05. Example05 is working fine even if merged with the other code. 

The following problem occurs:

If the board running Example05 lost it's power it's accessible again if power returns. In log I can see the following lines:
[ 5206] HomeKit: Starting server
[ 5210] HomeKit: Using existing accessory ID: xx:xx:xx:xx:xx:xx

If the bord running combined sketch lost it's power and power returns, it has to be paired again. Log says:
[ 7359] HomeKit: Starting server
[ 7362] HomeKit: Formatting HomeKit storage at 0x3fb000
[ 7458] HomeKit: Generated new accessory ID: yy:yy:yy:yy:yy:yy

What goes wrong and how can I prevent the sketch to forget the accessory ID?

SchneHa, 2020-11-02:
I know the reason but I have no solution.
I need to use an EEPROM library and I have to call EEPROM.begin(4096) to initialize. As a result the stored HK data are deleted and I have to pair again after power off.
Is it possible to read stored HK keys before and restore after initializing EEPROM? In used EEPROM library call of EEPROM.begin(4096) has to be done first.

UlrichMai, 2020-11-03:
@SchneHa, I had the same problem and found it difficult to change the way the HomeKit lib stores the settings. So I changed my initialization of the EEPROM lib. The constructor allows you to specify the memory segement to use:
`` EEPROM = EEPROMClass(0x3FB);    ``
I needed to change the memory layout in the LD file to add another 4k to the reserved eeprom space
```
/* Flash Split for 4M chips */
/* sketch @0x40200000 (~1019KB) (1044464B) */
/* empty  @0x402FEFF0 (~3052KB) (3125264B) */
/* spiffs @0x405FA000 (~0KB) (0B) */
/* eeprom @0x405FA000 (8KB) <<<<<*/
/* rfcal  @0x405FC000 (4KB) */
/* wifi   @0x405FD000 (12KB) */
```
hope this helps.

SchneHa, 2020-11-03:
@UlrichMai Thank you for the response but can you make it a little bit clearer. Where will I find the LD file and what exactly have I to change? Can you point me to an example?

Mixiaoxiao, 2020-11-03:
When you want to write something to flash, you should erase the full `sector` (in 4096B). This is done in EEPROM class internally if you use the EEPROM library. See the EEPROM library source code.
The EEPROM is just a `sector` in ESP.

So, if you want to use the EEPROM flash, you should: 
1. read all content of the full `sector` or some bytes larger than 1408 to a byte-buffer (HomeKit use 0~1408), e.g. 2048
2. save your bytes in this byte-buffer with the beginning address > 1408, e.g. 1536~2048 (DO NOT modify 0~1408)
3. write all byte-buffer to flash.

So, if you use the EEPROM library:
1. EEPROM.begin(2048)  the length should larger than 1408. Will read 0~2048 to a byte-buffer internally.
2. EEPROM.write(addr, your data)   addr should > 1408, so 0~1408 will not changed.
3. EEPROM.commit() or ~.end(). write back the full 2048 buffer to the flash.

SchneHa, 2020-11-03:
@Mixiaoxiao Yes, I know. But this doesn't answer my question. If I call EEPROM.begin(2048) all of the EEPROM content is gone, including HK date stored 0~1048. That means, that I have to pair my device again.

Mixiaoxiao, 2020-11-03:
> @Mixiaoxiao Yes, I know. But this doesn't answer my question. If I call EEPROM.begin(2048) all of the EEPROM content is gone, including HK date stored 0~1048. That means, that I have to pair my device again.

Wrong. Not caused by EEPROM.begin. There are only one `ESP.flashRead` call in EEPROM.begin.
Check your own code.

SchneHa, 2020-11-03:
@Mixiaoxiao Are we talking about the same library? I mean that one from Ivan Grokhotkov. Should I use another one? Which one do you recommend?

`void EEPROMClass::begin(size_t size) {
  if (size <= 0)
    return;
  if (size > SPI_FLASH_SEC_SIZE)
    size = SPI_FLASH_SEC_SIZE;

  size = (size + 3) & (~3);

  if (_data) {
    delete[] _data;
  }

  _data = new uint8_t[size];
  _size = size;

  noInterrupts();
  spi_flash_read(_sector * SPI_FLASH_SEC_SIZE, reinterpret_cast<uint32_t*>(_data), _size);
  interrupts();
}`

SchneHa, 2020-11-03:
Now it works! It was a bug in my code, indeed.
Thank you very much.

closed 2020-11-03 by SchneHa