Mixiaoxiao/Arduino-HomeKit-ESP8266 #81: Adding more thon one switch to homekit with one ESP8266

BenGore, 2020-11-09:
Hey guys,
first of all: Thank you so much for this beautiful possability to add ESP8266 to a homekit network!

I had a few tests with a single Relais and it worked pretty good.

Now I would like to know if it's possible to two (or more) Relais with one ESP8266 at the same time and how to do it.
This should be the same as a double switch. Unfortunately I'm not  so familiar with programming.

Many thanks in advance!
Best regards!

d4rkmen, 2020-11-09:
Hi, every lightbulb service will be displayed as a separate lightbulb.

BenGore, 2020-11-10:
> Hi, every lightbulb service will be displayed as a separate lightbulb.

Thank you for your quick reply. Is there any example for a lightbulb? 
And does it make sense to use a lightbulb for this? Is there no chance to create a device like a doubleswitch or a Smart Power Strip?

closed 2020-11-26 by BenGore