Mixiaoxiao/Arduino-HomeKit-ESP8266 #82: How can I add a physical button to switch example?

greg21greg, 2020-11-13:


Mixiaoxiao, 2020-11-13:
Ask Google.

greg21greg, 2020-11-13:
I tried that already :( I was hoping someone could help me out here.

JGConcepcion, 2020-12-12:
Were you able to make it work? I'm trying to make it work

pratik019, 2020-12-14:
I tried it, within switch example there is 3 lines of commented code to submit led pin status to ios app. from line 77 to 79.

but when i did that system started going into reset mode continiously exception(0), i would love help with it.

JGConcepcion, 2020-12-14:
I was able to make it work already but it involved the use of another library which is the ESPbutton. I added this piece of code (See attached image) into the my_homekit_setup() and I got no error or whatsoever. 
[image]

yepkevin612, 2020-12-23:
@JGConcepcion con you share your code?

Varogkorn, 2021-02-15:
I used  **[Acebutton Libray](https://github.com/bxparks/AceButton)** to work with **Push button Switch** with **INPUT_PULLUP**

- [ ] When you turn ON the swich The LED will on and it update state in Home App also same like when you switch OFF.

### **Edit program in Example02_Switch.ino**

- First,Add this before void setup()
	```C
	 #include <AceButton.h>
           using namespace ace_button;
           const int BUTTON_PIN = 14; //edit your switch pin here.
           AceButton button(BUTTON_PIN);
          void handleEvent(AceButton*, uint8_t, uint8_t);
          int lock_state;
	```

- Second,In void setup() add
	```C
  pinMode(BUTTON_PIN, INPUT_PULLUP);
  button.setEventHandler(handleEvent);
	```
- Third,Add button.check(); in your void loop() like this.
```C
void loop() {
	my_homekit_loop();
        button.check(); //ADD THIS.
	delay(10);
}
```

- Last, Add the switch event funtion(Like below) at the below of progarm.
```C
void handleEvent(AceButton* /* button */, uint8_t eventType,
    uint8_t /* buttonState */) {
  switch (eventType) {
    case AceButton::kEventPressed:
      if(lock_state==0){
        cha_switch_on.value.bool_value = true;
        bool on = cha_switch_on.value.bool_value;
        Serial.println("LED:ON(Force)");
        digitalWrite(PIN_SWITCH, HIGH);
        homekit_characteristic_notify(&cha_switch_on, cha_switch_on.value);
        lock_state = 1;
        }
      break;
    case AceButton::kEventReleased:
      if(lock_state==1){
        cha_switch_on.value.bool_value = false;
        bool on = !cha_switch_on.value.bool_value;
        Serial.println("LED:OFF(Force)");
        digitalWrite(PIN_SWITCH, LOW);
        homekit_characteristic_notify(&cha_switch_on, cha_switch_on.value);
        lock_state = 0;
      }
      break;
  }
}
```

yepkevin612, 2021-02-24:
@Varogkorn thank you so much，i still cannot turn off ，when i hold the button there is off，when i released my hand the switch continous turn on.

alexxxpol, 2021-03-02:
Hi!
void handleEvent(AceButton* /* button */, uint8_t eventType,
    uint8_t /* buttonState */) {
      //Serial.println('lock_state');
  switch (eventType) {
    case AceButton::kEventPressed:
      if(lock_state==1){
        cha_switch_on.value.bool_value = true;
        bool on = cha_switch_on.value.bool_value;
        //Serial.println("LED:ON(Force)");
        digitalWrite(PIN_SWITCH, HIGH);
        homekit_characteristic_notify(&cha_switch_on, cha_switch_on.value);
        lock_state = 0;
        }
        else {
        cha_switch_on.value.bool_value = false;
        bool on = !cha_switch_on.value.bool_value;
        //Serial.println("LED:OFF(Force)");
        digitalWrite(PIN_SWITCH, LOW);
        homekit_characteristic_notify(&cha_switch_on, cha_switch_on.value);
        lock_state = 1;
      }  
      break;
    case AceButton::kEventReleased:
    break;
  }
}

yepkevin612, 2021-03-03:
@alexxxpol Thank you , it work nice.

alexandrerochasalles, 2021-04-09:
How to change the HIGH to LOW position on home app?? I mean, even the exemple works inverse for me, it is on when off in home app, and off when it's on in my home app.

david19610, 2021-04-09:
> I'm using a Wemos d1 mini v3

Hello,

Pleas in function swich_on_set(), change the digitalWrite(2, switch_1_power) to digitalWrite(2, !(switch_1_power)).

void switch_1_on_set(homekit_value_t value) {
  if (value.format != homekit_format_bool) { printf("Invalid on-value format: %d\n", value.format); return;}
  switch_1_power = value.bool_value;
    digitalWrite(2, switch_1_power);
  //led_update();
}

Pozdrawiam/Mit freundlichen Grüssen 
Dawid Rosak
TEL. +48 609105069

alexandrerochasalles, 2021-04-09:
> Hello, Pleas in function swich_on_set(), change the digitalWrite(2, switch_1_power) to digitalWrite(2, !(switch_1_power)). void switch_1_on_set(homekit_value_t value) { if (value.format != [...]

I can't find the function that you are talking about. 
Can you please give me the full code? Because for the switch function I'm using as the switch example:

`void cha_switch_on_setter(const homekit_value_t value) {
	bool on = value.bool_value;
	cha_switch_on.value.bool_value = on;	//sync the value
	LOG_D("Switch: %s", on ? "ON" : "OFF");
	digitalWrite(PIN_SWITCH, on ? LOW : HIGH);
}

void my_homekit_setup() {
	pinMode(PIN_SWITCH, OUTPUT);
	digitalWrite(PIN_SWITCH, HIGH);

	cha_switch_on.setter = cha_switch_on_setter;
	arduino_homekit_setup(&config);

}`

david19610, 2021-04-09:
Hello,

you probably need to refresh status of the button, see the highlighted text
in yellow

Pozdrawiam
Dawid Rosak

> Hello, Pleas in function swich_on_set(), change the digitalWrite(2,
> switch_1_power) to digitalWrite(2, !(switch_1_power)). void
> switch_1_on_set(homekit_value_t value) { if (value.format !=
> homekit_format_bool) { printf("Invalid on-value format: %d\n",
> value.format); return;} switch_1_power = value.bool_value; digitalWrite(2,
> switch_1_power); //led_update(); } Pozdrawiam/Mit freundlichen Grüssen
> Dawid Rosak TEL. +48 609105069
> … <#m_2188757451150947713_>
>
> I can't find the function that you are talking about.
> Can you please give me the full code? Because for the switch function I'm
> using as the switch example:
 // access your HomeKit characteristics defined in my_accessory.c
extern "C" homekit_server_config_t config;
extern "C" homekit_characteristic_t cha_switch_on;

void cha_switch_on_setter(const homekit_value_t value) {
>
>         bool on = value.bool_value;
>
> 	cha_switch_on.value.bool_value = on;	//sync the value
>
>           //update current status of Button on Apple Home app
          homekit_characteristic_notify(&cha_switch_on,
cha_switch_on.value);

alexandrerochasalles, 2021-04-10:
@david19610 Thank you so much for the help!!

madmacks59, 2021-12-31:
Search for a project named "4 Relay & Switch" by Sachin Soni (aka techiesms). He has several projects that use physical switches and relays and I've got it working fine with HomeKit on a couple of different devices.

DenisMaldonadoC, 2022-11-22:
The problem with the Techiesms project is that when you interact with the HomeKit App, you have to doble click the physical switch. I made the project a few months ago and I’m still dealing with that problem.