Mixiaoxiao/Arduino-HomeKit-ESP8266 #86: Example05_WS2812_Neopixel 的疑问

fox007ggr, 2020-11-27:
void setup() {
	Serial.begin(115200);
	wifi_connect(); // in wifi_info.h
  
  
  pixels.begin(); 
  for(int i = 0; i < NUMPIXELS; i++)
  {
    pixels.setPixelColor(i, 5, 0, 0);
  }
  pixels.show();

  my_homekit_setup();
  
  delay(1000);
	rgb_colors[0] = 255;
  rgb_colors[1] = 255;
  rgb_colors[2] = 255;
	
}

my_homekit_setup(); //这条语句执行后

...
pixels.show(); //以后在执行这个WS2812 不会发生任何改变

my_homekit_setup();  //如果不执行这条语句 

pixels.show(); //是可以正常使用的

Mixiaoxiao, 2020-11-27:
按160MHz编译试一下

fox007ggr, 2020-11-27:
谢谢，可以了，在哪儿看到的这个参数

closed 2020-11-27 by fox007ggr

Ondrados, 2020-12-28:
@fox007ggr Hello, can you please tell me how did you solve it?

Ondrados, 2020-12-28:
@fox007ggr Nevermind. Its in Arduino IDE under Tools->CPU frequency