Mixiaoxiao/Arduino-HomeKit-ESP8266 #87: 定义一个水阀

fox007ggr, 2020-11-30:
现在想定义一个水阀

/**
 Defines that the accessory contains a (water) valve.
 
 Required Characteristics:
 - ACTIVE
 - IN_USE
 - VALVE_TYPE
 
 Optional Characteristics:
 - NAME
 - SET_DURATION
 - REMAINING_DURATION
 - IS_CONFIGURED
 - SERVICE_LABEL_INDEX
 - STATUS_FAULT
 */
#define HOMEKIT_SERVICE_VALVE HOMEKIT_APPLE_UUID2("D0")

但是实现部分

HOMEKIT_ACCESSORY(.id=3, .category=**这里该怎么定义呢，文档里面没找到**, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
         ...
       }

Mixiaoxiao, 2020-11-30:
category一般取 xxxx_others即可

Chehadeej, 2022-10-09:
Did it work? Can you please show me how you did it? I m facing the same issue