Mixiaoxiao/Arduino-HomeKit-ESP8266 #88: Outlet and temp sensor not possible...

thncode, 2020-11-30:
@Mixiaoxiao I came to a strange point with my accessories: I do have a working outlet accessory in two flavors: as an energy meter and as a simple outlet, both even working with eve history features. But whenever I want to add a temperature accessory, either as a characteristic or as part of a bridge, HomeKit says "not compatible". The bridge example works though, and I do not see a difference in the respective code parts. Even all the other sensors (humidity, motion, ...) do work, but not with temperature.
Is there any place in arduino_homekit_server.cpp were I could place  some checkpoints to see why it refuses the temperature characteristic?

THANKS!

Mixiaoxiao, 2020-11-30:
Double check your temperature-characteristic's value. Wrong format or value is out of range?

thncode, 2020-11-30:
Yes, that was the problem. Thx a lot, I never would have though there is a connection...

closed 2020-11-30 by thncode