Mixiaoxiao/Arduino-HomeKit-ESP8266 #91: Brightness control

RobinH6, 2020-12-06:
Anyone nows how to setup brightness control?
[image]

sircuri, 2020-12-14:
Show us some code that you have already. We might be able to help.

RobinH6, 2020-12-22:
I did it :D

```
#include <Arduino.h>
#include <arduino_homekit_server.h>
#include "wifi_info.h"

#define LOG_D(fmt, ...)   printf_P(PSTR(fmt "\n") , ##__VA_ARGS__);

bool is_on = false;
float current_brightness =  100.0;

void setup() {
  Serial.begin(115200);
  wifi_connect(); // in wifi_info.h
  my_homekit_setup();
}

void loop() {
  my_homekit_loop();
  delay(10);
}

//==============================
// HomeKit setup and loop
//==============================

// access your HomeKit characteristics defined in my_accessory.c

extern "C" homekit_server_config_t accessory_config;
extern "C" homekit_characteristic_t cha_on;
extern "C" homekit_characteristic_t cha_bright;

static uint32_t next_heap_millis = 0;

const int LED = D4;

void my_homekit_setup() {
  
  pinMode(LED, OUTPUT);
  cha_on.setter = set_on;
  cha_bright.setter = set_bright;

  arduino_homekit_setup(&accessory_config);
}

void set_on(const homekit_value_t v) {
    bool on = v.bool_value;
    cha_on.value.bool_value = on; //sync the value

    if(on) {
        is_on = true;
        Serial.println("On");
    } else  {
        is_on = false;
        Serial.println("Off");
    }
}

void set_bright(const homekit_value_t v) {
    Serial.println("set_bright");
    int bright = v.int_value;
    cha_bright.value.int_value = bright; //sync the value

    current_brightness = bright;
    Serial.println(bright);
    updateBrightness();
}

void updateBrightness() {
  if(is_on)
  {
    int b = map(current_brightness,0, 100,75, 255);
      Serial.println(b);
      analogWrite(D4,b);
   }
   
  else if(!is_on) //lamp - switch to off
  {
      Serial.println("is_on == false");
      analogWrite(D4, 0);
      
  }
}

void my_homekit_loop() {
  arduino_homekit_loop();
  const uint32_t t = millis();
  if (t > next_heap_millis) {
    // show heap info every 5 seconds
    next_heap_millis = t + 5 * 1000;
    LOG_D("Free heap: %d, HomeKit clients: %d",
        ESP.getFreeHeap(), arduino_homekit_connected_clients_count());

  }
}
```

but how can i set a smother brightness range for example to 0-1024?
Have a nice Day

JustBertC, 2020-12-22:
I would try changing this:
`int b = map(current_brightness,0, 100,75, 255);`

to this:
`int b = map(current_brightness,0, 100,0, 1023);`

RobinH6, 2020-12-22:
thats nice!!
how can i set the start brightness? When i set in my_accessory.h the brightness to 50 the app show but the led brightness is 100% (int 1023).
Also I have add `updateBrightness();` in void set_on because set_on had no function without that. code seems so:
```
void set_on(const homekit_value_t v) {
    bool on = v.bool_value;
    cha_on.value.bool_value = on; //sync the value

    if(on) {
        is_on = true;
        Serial.println("On");
        updateBrightness();
        
    } else  {
        is_on = false;
        Serial.println("Off");
        updateBrightness();
        
        
    }
}
```

JustBertC, 2020-12-23:
You could add this line into the setup routine after pinMode:
`analogWrite(D4, 512);`

That will force the LED to be at 50% brightness also. There may be a way to get the esp to sync the value from the home app to initialise it properly but I haven't looked into the code that deeply yet.

mstaack, 2022-03-12:
@RobinH6 do you mind posting your accessory config? mine does not work:

```
/*
 * my_accessory.c
 * Define the accessory in C language using the Macro in characteristics.h
 *
 *  Created on: 2020-05-15
 *      Author: Mixiaoxiao (Wang Bin)
 */

#include <homekit/homekit.h>
#include <homekit/characteristics.h>

void my_accessory_identify(homekit_value_t _value) {
	printf("accessory identify\n");
}

homekit_characteristic_t cha_on = HOMEKIT_CHARACTERISTIC_(ON, false);
homekit_characteristic_t cha_name = HOMEKIT_CHARACTERISTIC_(NAME, "Dimmer");
homekit_characteristic_t cha_bright = HOMEKIT_CHARACTERISTIC_(BRIGHTNESS, 50);

homekit_accessory_t *accessories[] = {
    HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_lightbulb, .services=(homekit_service_t*[]) {
        HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]) {
            HOMEKIT_CHARACTERISTIC(NAME, "Dimmer"),
            HOMEKIT_CHARACTERISTIC(MANUFACTURER, "Arduino HomeKit"),
            HOMEKIT_CHARACTERISTIC(SERIAL_NUMBER, "0123456"),
            HOMEKIT_CHARACTERISTIC(MODEL, "ESP8266/ESP32"),
            HOMEKIT_CHARACTERISTIC(FIRMWARE_REVISION, "1.0"),
            HOMEKIT_CHARACTERISTIC(IDENTIFY, my_accessory_identify),
            NULL
        }),
		HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
			&cha_on,
			&cha_name,
      &cha_bright,
			NULL
		}),
        NULL
    }),
    NULL
};

homekit_server_config_t config = {
		.accessories = accessories,
		.password = "111-11-111"
};
```